adafruit/circuitpython #7377: wifi.radio: provide "connected" and "access point is up" boolean properties

dhalbert, 2022-12-23:
I have seen many people asking how to find out whether a `wifi.Radio connection` is up or not. The method usually described is to try to fetch the current IP address. If that fails, the connection is not up.

I think more straightforward boolean properties would be helpful in the API, even if they are bit redundant.

### Station mode
#### Existing API

- `wifi.Radio.connect()`: connects to a specified network as a Station.
- `wifi.Radio.start_station()`: start running as a Station.
- `wifi.Radio.stop_station()`: stop running as a Station.

#### Additional proposed API

- `wifi.Radio.connected: bool`: True if up and running as a Station. Read-only
OR
`- wifi.Radio.station_active: bool`
OR
- `wifi.Radio.station_up: bool`

### Access Point mode
#### Existing API

- `wifi.Radio.start_ap()`
- `wifi.Radio.stop_ap()`

#### Additional proposed API
- `wifi.radio.ap_active: bool`
OR
- `wifi.radio.ap_up: bool`

Any comments on the idea and the names? In the current documentation, `Station` and `Access Point` are capitalized, not sure why.

This is quite simple to implement.

anecdata, 2022-12-23:
I like `wifi.Radio.connected`, it's direct and reflextive of `connect()`. The start/stop station are under-the-hood things that most people probably don't use, but in station (and moreso in AP mode) they can be useful to change modes and change some properties before starting the mode back up.

Maybe `started` for AP since `start` is the verb?

dhalbert, 2022-12-24:
@anecdata thanks for the comments. If we were starting from scratch, the API might be more symmetric, but I would not be inclined to change that now.

> Maybe `started` for AP since `start()` is the verb?

It looks to me like the CYW43 can be in an intermediate state where the AP has "started to start" but there is no IP yet. I'm thinking `ap_started()` may not mean "ap ready right now". So I think "ap active" or "ap up" might make it clearer that it's up and ready to use.

RetiredWizard, 2022-12-24:
I also like the `wifi.radio.connected` option, although, I'd offer a suggested change to `wifi.radio.is_connected` since adafruit_esp32spi.ESP_SPIcontrol already uses that key word.

tannewt, 2023-01-04:
I agree with @anecdata and would rather not use `up`. It is a network term that makes no sense without that background.

dhalbert, 2023-01-04:
I did most of a PR when I opened an issue. Will try to finish it soon.

anecdata, 2023-01-10:
Could this play into a possible major version API change to allow async network actions, like in the present case: async wifi connection (rather than blocking for several seconds)?

closed 2023-04-01 by dhalbert

cn-schuh, 2023-05-03:
Great, many Thanks to You. Just started to rebuild my esp32-s2 wifi connection function.

PR #7823 (2023-04-01): Add wifi.radio.connected, wifi.radio.ap_active
- Fixes #7377.

- Add `wifi.Radio.connected` read-only property.
- Add `wifi.Radio.ap_active` read-only property.
- Rework some `wifi` documentation slightly.

Tested on Pico W and Metro ESP32-S2.