adafruit/circuitpython #7379: Async Pin interupts not working on RPI Pico?

schwa, 2022-12-25:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Pimoroni Tiny 2040 (2MB) with rp2040
```

### Code/REPL

```python
async def catch_interrupt(pin):
    """Print a message when pin goes low."""
    with countio.Counter(pin) as interrupt:
        while True:
            if interrupt.count > 0:
                interrupt.count = 0
                print("interrupted!")
            # Let another task run.
            await asyncio.sleep(0)

async def main():
    interrupt_task = asyncio.create_task(catch_interrupt(board.GP26))
    await asyncio.gather(interrupt_task)

asyncio.run(main())
```

### Behavior

I'm following the example on https://learn.adafruit.com/cooperative-multitasking-in-circuitpython-with-asyncio/handling-interrupts on a Pimoroni Tiny 2040. I have a momentary button hooked up to GP26 and have modified the source code from the example.

When I run I get:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 62, in <module>
  File "asyncio/core.py", line 297, in run
  File "asyncio/core.py", line 261, in run_until_complete
  File "asyncio/core.py", line 246, in run_until_complete
  File "code.py", line 59, in main
  File "asyncio/funcs.py", line 113, in gather
  File "asyncio/funcs.py", line 108, in gather
  File "asyncio/core.py", line 246, in run_until_complete
  File "code.py", line 48, in catch_interrupt
RuntimeError: Pin must be on PWM Channel B

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "code.py", line 62, in <module>
  File "asyncio/core.py", line 297, in run
  File "asyncio/core.py", line 261, in run_until_complete
  File "asyncio/core.py", line 246, in run_until_complete
  File "code.py", line 59, in main
  File "asyncio/funcs.py", line 113, in gather
  File "asyncio/funcs.py", line 108, in gather
  File "asyncio/core.py", line 246, in run_until_complete
  File "code.py", line 48, in catch_interrupt
RuntimeError: Pin must be on PWM Channel B

Code done running.

Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Pimoroni Tiny 2040 (2MB) with rp2040
```

I'm just trying to get a simple button working - no PWM involved.

### Additional information

Error is thrown here I think: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/countio/Counter.c#L20

Neradoc, 2022-12-25:
Hi, `countio` is not technically an interrupt module, it counts edges. On RP2040 it uses PWM for that, and due to hardware limitations it requires a pin on a PWM channel B. Unfortunately the pico Learn Guide doesn't list which pins are valid for that, which could be a good addition. (The information can be found in the RP2040 datasheet section 4.5.2).
As it turns out it's odd number pins.

To monitor a pin, I would rather advise using the `keypad` module, which regularly scans pins in the background and provides a list of press/release events that can be read in python code:
https://learn.adafruit.com/key-pad-matrix-scanning-in-circuitpython

Here is some similar code that uses the keypad module:
```py
import asyncio
import keypad

async def catch_button(pin):
    """Print a message when pin goes low."""
    with keypad.Keys((pin,), value_when_pressed=False, pull=True) as keys:
        while True:
            if event := keys.events.get():
                if event.pressed:
                    print(f"Button pressed !")
            await asyncio.sleep(0)

async def main():
    interrupt_task = asyncio.create_task(catch_button(board.GP26))
    await asyncio.gather(interrupt_task)

asyncio.run(main())
```

_For further questions and community help I suggest going to the [Adafruit discord server](https://adafru.it/discord) or the [support forums](https://forums.adafruit.com/)._

schwa, 2022-12-25:
All makes sense. I’d be fine with closing this issue then unless it would be more useful as a documentation issue. Thank you all.

dhalbert, 2022-12-25:
I agree we could say something about the pin restrictions in the ReadTheDocs documentation. Turning this into a doc issue.

todbot, 2022-12-26:
As a note about quickly finding GPIO capabilities on the Pico, I really like https://pico.pinout.xyz/ with the “Advanced” box ticked

PR #8104 (2023-06-22): note RP2040 countio limitations
- Fixes #7379.

Documentation update only.
closed 2023-06-22 by dhalbert