adafruit/circuitpython #7380: CIRCUITPY_WEB_API_PORT being ignored - ProS3 / ESP32S3

grbd, 2022-12-26:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; ProS3 with ESP32S3
(Using latest U2F version with latest bootloader)
```

### Code/REPL

```python
`settings.toml`

CIRCUITPY_WIFI_SSID="GBD-TPLINK-Ext-2.4GHz"
CIRCUITPY_WIFI_PASSWORD="secretpassword"
CIRCUITPY_WEB_API_PASSWORD="password"
CIRCUITPY_WEB_API_PORT=3000
```
```

### Behavior

It looks as if CIRCUITPY_WEB_API_PORT is being ignored.
The webapi works but it just defaults to port 80 every time.

Tried a lower value of 70 for the port but it made no difference
Tried in and out of quotes (should be out of quotes) but made no difference
I kept my ssid real in case there's something in it messing with the setup, but not posting my password for obvious reasons, although that only contains letters and numbers but no special characters

jepler, 2022-12-27:
I tested with an S3-based feather and still didn't reproduce the problem. I tried 70, 81, 3000, and 8080 as alternate port numbers.  I tested with the CIRCUITPY_WEB_API_PORT line being first, middle, and last.

Let's eliminate the possibility of strange control codes in your settings.toml.  Can you please run this in the repl, and then redact the wifi password in the printed string before pasting it into a comment with \`\`\`backticks\`\`\`?
```
>>> open("/settings.toml", "rb").read()
b'CIRCUITPY_WIFI_PASSWORD="redacted"\nCIRCUITPY_WIFI_SSID="NETGEAR63"\nCIRCUITPY_WEB_API_PASSWORD="also redacted"\nCIRCUITPY_WEB_API_PORT=81'
```

grbd, 2022-12-27:
Just tried `adafruit-circuitpython-unexpectedmaker_pros3-en_GB-20221227-785ebe1.uf2` which was released at 2022-12-27T19:18:42.000Z
Not sure if it picked up the recent pull request

Used the following settings
`settings.toml`
```
CIRCUITPY_WIFI_SSID="GBD-TPLINK3-Ext-2.4GHz"
CIRCUITPY_WIFI_PASSWORD="****"
CIRCUITPY_WEB_API_PORT=3000
CIRCUITPY_WEB_API_PASSWORD="password"
#
#

```

Reset the device
Looking at the IP address assigned
```
import wifi
print("my IP addr:", wifi.radio.ipv4_address)
# Shows 192.168.111.114
```

Tested with

  * http://192.168.111.114/ <- Shows the welcome to web api page
  * http://192.168.111.114:3000/ <- No response

Tried reading the config
```
open("/settings.toml", "rb").read()
# Result:
b'CIRCUITPY_WIFI_SSID="GBD-TPLINK3-Ext-2.4GHz"\r\nCIRCUITPY_WIFI_PASSWORD="****"\r\nCIRCUITPY_WEB_API_PORT=3000\r\nCIRCUITPY_WEB_API_PASSWORD="password"\r\n#\r\n#\r\n'
```

grbd, 2022-12-27:
Also Tried
```
CIRCUITPY_WEB_API_PORT=81
CIRCUITPY_WEB_API_PASSWORD="password"
CIRCUITPY_WIFI_SSID="GBD-TPLINK3-Ext-2.4GHz"
CIRCUITPY_WIFI_PASSWORD="*****"
```

```
open("/settings.toml", "rb").read()
b'CIRCUITPY_WEB_API_PORT=81\r\nCIRCUITPY_WEB_API_PASSWORD="password"\r\nCIRCUITPY_WIFI_SSID="GBD-TPLINK3-Ext-2.4GHz"\r\nCIRCUITPY_WIFI_PASSWORD="****"\r\n'
```

but same result

grbd, 2022-12-28:
I think I figured out the cause
I noticed the following on the serial console if I could capture it early enough after a reset
```
An error occurred while retrieving 'CIRCUITPY_WEB_API_PORT':
Invalid byte '\r'
```

and I was like aha it's windows line feeds.
Switched my editor to save unix '\n' instead of windows '\r\n' line feeds and it works
Doesn't seem to impact the other settings just that one CIRCUITPY_WEB_API_PORT
so I'm guessing the other settings are filtering out the windows line feeds some how.

PR #7395 (2022-12-28): Fix getenv_int handling of "dos-style" newlines
I think this closes #7380 ; closes #7394.
closed 2022-12-28 by dhalbert