adafruit/circuitpython #7383: Wifi Issues when connecting, with multiple stations on the same channel ESP32-S3

grbd, 2022-12-26:
### CircuitPython version

```python
Both
Adafruit CircuitPython 7.3.3 on 2022-08-29; ProS3 with ESP32S3
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; ProS3 with ESP32S3
```

### Code/REPL

```python
Hi,
I'm not sure if this is a circuitpython issue or something related to the esp-idf that circuitpython is using
But I recently ran into a problem similar to this one

  * https://github.com/adafruit/circuitpython/issues/4373

From what I can tell if there's multiple stations available on a single channel
Then the esp seems to have problems connecting to one of them

## Example - Scanning

So running the following from home to get a list of access points

import wifi
for network in wifi.radio.start_scanning_networks():
    print("\t%s\t\tRSSI: %d\tChannel: %d" % (str(network.ssid, "utf-8"),
            network.rssi, network.channel))
wifi.radio.stop_scanning_networks()

I end up with the following
```
GBD-TPLINK-Ext-2.4GHz           RSSI: -47       Channel: 3
GBD2_2.4                        RSSI: -79       Channel: 3
GBD-TPLINK3-Ext-2.4GHz          RSSI: -74       Channel: 2
```

  * The first one is a wifi booster sitting in the same room I'm in so the signal strength is pretty strong
  * The second is the main wifi router, that's in another room in the house
  * The third is another wifi booster sitting in the garage far away

## Example Connecting

Trying to connect with
```
print(wifi.radio.connect(ssid="GBD-TPLINK-Ext-2.4GHz",password="passwordhere",timeout=60))
```

  * Typically for the first two 90% of the time when trying a connection I'll end up with a `ConnectionError: Unknown failure 205` assuming I extend the timeout to 60 or longer.
  * For the third which is on a different channel the 205 error never shows up, the signal is a bit weak so I might get "No network with that ssid" but I can connect usually on about 2 attempts
  * I can increase the odds a bit of connecting to the access point in the same room (the first one) by placing the device right next to it.

This seems to suggest some sort of interfernce going on when there's more than one access point on the same channel.
I tend to have my wifi boosters setup with different ssid's so I can tell what I'm connecting to.
Both the boosters are the same model
So I'm wondering if this might be an ESP-S3 / ESP-S2 bug within the esp-idf libs

I've tried this on the latest V7 and V8 firmwares with the same result
Also tried specifying the bssid in the options but that results in the same 205 error
```

### Behavior

As Above

anecdata, 2022-12-27:
Your wifi scan doesn't show it, but are there a lot of other wifi devices in the vicinity that could be causing interference? Are the routers choosing their channels, or are those manually selected? Channels 2 and 3 are adjacent, and also have frequency overlap with channels1 and 6. Channels 1, 6, and 11 are the only combination that have no overlap.

There was a comment on another issue that the ESP32-S3 has a hardware bug that reduces the signal strength on the lower channel numbers:
https://github.com/adafruit/circuitpython/issues/6791#issuecomment-1264938562
Does it help if you push those channels to higher numbers?

grbd, 2022-12-27:
Ok a bit more investigation
I did another scan just now
```
        Virgin Media            RSSI: -75       Channel: 6      Bssid: b'Z\xd3C1\xd2\xe1'
        BT-7CAG8T               RSSI: -90       Channel: 1      Bssid: b'\xecl\x9a\x1cq\x80'
        PLUSNET-6N79_Ext                RSSI: -93       Channel: 11     Bssid: b"\xc0V'\xa5`\xda"
        GBD-TPLINK-Ext-2.4GHz           RSSI: -47       Channel: 3      Bssid: b'\xcc2\xe58\xa8d'
        GBD-TPLINK3-Ext-2.4GHz          RSSI: -74       Channel: 3      Bssid: b'\xe8H\xb8c\xd9\xd9'
        GBD2_2.4                RSSI: -77       Channel: 3      Bssid: b'\xfc4\x97\xd3\x16p'
        OURHOUSE                RSSI: -89       Channel: 2      Bssid: b'\xcc2\xe5[D\x06'
```

The three GBD ones are all mine and all of them appear to be on channel 3
Unfortunatley I can't pick the channel number for the tplink extenders, although I am going to experiment with the main wifi router channnel later on

If I try GBD2_2.4 or GBD-TPLINK-Ext-2.4GHz then instant error
If I try GBD-TPLINK3-Ext-2.4GHz then it works assuming it can get a connection since the power is distant
this is kind of weird since they're all showing on the same channel number

I'm noticing that GBD-TPLINK3-Ext-2.4GHz is actually connected to GBD-TPLINK2-Ext-2.4GHz which isn't visible by the ESP and that's then connecting to GBD2_2.4 the main wifi
so ESP -> GBD-TPLINK3-Ext-2.4GHz -> (2.4Ghz but not 5Ghz) GBD-TPLINK2-Ext-2.4GHz -> (2.4/5Ghz) Main wifi

I'm going to have to experiment with this a bit more to try and see if I can figure out a pattern

grbd, 2022-12-27:
just as a follow up
 
  * my main wifi router is a Asus GT-AX11000
    latest firmware 3.0.0.4.386_49599-g8352df7
  * the wifi boosters are all the same, TPLink RE650 V1, latest firmware 20201112

Tried disabling 5Ghz on one of the boosters I couldn't connect to but made no difference
What I think I'm seeing at the moment is that if a connection is made via ESP -> Booster -> Booster -> Router, then I think it works for some reason.

Next point of call is to try altering some of the wifi settings such as picking a different channel on the main router

grbd, 2022-12-28:
I think I've managed to solve this one
Going to close it down since if this is an issue and not wifi interference (which it could be) it's probably more of a hardware / esp-idf issue
that being said it maybe useful for others to know

## Fix - set "Channel bandwidth" to "40Mhz"

The short answer is that in my wifi router settings
changing the "Channel bandwidth" setting from "20/40Mhz" to "40Mhz" seems to allow the esp32-s3 to connect on the lower channel numbers without issue

## Results for "Channel bandwidth" set to "20/40Mhz" on router

If "Channel bandwidth" is set to it's default of "20/40Mhz" then the end results for connecting to different channels are

  * ch1 (above only) - 205 Error
  * ch2 (above only) - 205 Error
  * ch3 (above only) - 205 Error
  * ch4 (above only) - 205 Error
  * ch5 (below) - works
  * ch5 (above) - works but takes a while
  * ch6 (below) - 205 Error
  * ch6 (above) - 205 Error
  * ch7 (below) - 205 Error
  * ch7 (above) - works
  * ch8 (below) - works - *
  * ch8 (above) - works
  * ch9 (below) - works
  * ch9 (above) - works
  * ch10 (below only) - works
  * ch11 (below only) - works
  * ch12 (below only) - intermittent 205 / works sometimes
  * ch13 (below only) - works

The below / above setting is something called "Extension Channel" some channels have the option of below some above and some you can pick ether
It seems as if the esp device has issues with the lower channels when trying to use 20Mhz instead of 40Mhz bandwidth
maybe a default needs to be altered within the esp-idf sources some place

closed 2022-12-28 by grbd