adafruit/circuitpython #7385: AP mode not working on ESP32-S2 in latest build. No AP started, without raising an error.

louisprp, 2022-12-27:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Waveshare ESP32-S2-Pico with ESP32S2
```

### Code/REPL

```python
import wifi

# at this point, the ESP32-S2 is running as a station (init default), and if desired can connect to any AP
wifi.radio.enabled = False  # turns wifi off, mode is retained or can be changed while not enabled

print("Stopping the (default) station...")
wifi.radio.stop_station()  # now the device is in NONE mode, neither Station nor Access Point

print("Starting AP...")
wifi.radio.start_ap("Bob", "12345678")

wifi.radio.enabled = True  # turns wifi back on

print(wifi.radio.mac_address_ap)
print(wifi.radio.enabled)
```

### Behavior

Stopping the (default) station...
Starting AP...
b'\x84\xf7\x03\xd9\xf2;'
True

### Description

The device does not create an access point, even when a channel is manually set, or the mode is set to open. I have tried out different combinations of disabling and enabling the wifi and stopping the stations, but with no success. It should be noted that connecting to an AP is working perfectly fine. I am not sure as to what is causing this, as I was referring to a previous issue https://github.com/adafruit/circuitpython/issues/4246 when trying to implement the code. From the issue, I concluded the functionality should be working on the latest version and has already been implemented. Might the board I am using have to do with the issue? 
I am not sure what I am doing wrong here, as there also does not seem to be an error raised, even when catching for exceptions.

dhalbert, 2022-12-27:
Are you running with the wifi workflow set up (via `settings.toml`) or not?

I tried this on a Metro ESP32-S2 with beta.6, with using `.enabled` and without, and it worked for me. I saw the network advertised on my phone. I don't think you need bother to do the `.enabled = False` and then re-enabling.

louisprp, 2022-12-27:
If you mean the `CIRCUITPY_WIFI_PASSWORD` and `CIRCUITPY_WIFI_SSID`, from the environment, I made sure to clear them out before trying. I have also tried not to disable and re-enable the wifi, but confusingly, there seems to be no change in the behaviour.

louisprp, 2022-12-27:
I have just updated to the latest version of CircuitPython and it seems to work fine now. It could have been just an error with my code. Thank you for the quick help!

closed 2022-12-27 by louisprp