adafruit/circuitpython #7386: Initialising Wifi Manager within an object __init__ crashed circuitpython

markchapman3216, 2022-12-27:
### CircuitPython version

```python
Adafruit circuitpython 7.3.3 on Feather M4 express
```

### Code/REPL

```python
import carrierboardv2 as cb
import time
import rtc
import board
import busio
from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_wifimanager
import adafruit_esp32spi.adafruit_esp32spi_socket as socket

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

import helperfunctions as hf

class Controller_Board_V2:
    def __init__(self, HAS_DS18B20):
        self.clock = rtc.RTC();
        if(HAS_DS18B20):
            from adafruit_onewire.bus import OneWireBus
            from adafruit_onewire.bus import OneWireAddress
            import adafruit_ds18x20
            self.ow_bus = OneWireBus(cb.OneWirePin)
            
            try:
                from secrets import identities
            except ImportError:
                print("Device identities are kept in secrets.py, please add them there!")
                raise
            
            self.maintempsense_addr = OneWireAddress(identities["maintempsensor"])
            self.ds18b20 = adafruit_ds18x20.DS18X20(self.ow_bus, self.maintempsense_addr)
            self.ds18b20.resolution = 12
            self.temp_now = self.ds18b20.temperature
            print('Temperature: {0:0.3f} °C'.format(self.temp_now))
        ### Display Initialisation ###
        import terminalio
        import displayio
        from adafruit_display_text import label
        from adafruit_bitmap_font import bitmap_font
        import adafruit_displayio_ssd1306
        displayio.release_displays()
        self.display_bus = displayio.I2CDisplay(cb.i2c, device_address=0x3C)
        self.display = adafruit_displayio_ssd1306.SSD1306(self.display_bus, width=128, height=32)
        self.splash = displayio.Group()
        self.display.show(self.splash)
        self.font = bitmap_font.load_font("/fonts/Arial-Black-20.bdf")
        # Thermometer area
        text = "xx.x`C"  # Note, font has been hacked to turn ` into degree symbol
        text_area = label.Label(self.font, text=text, color=0xFFFF00, x=2, y=11)
        self.splash.append(text_area)

        # Clock area
        clock_text = "12:00"
        clock_area = label.Label(terminalio.FONT, text=clock_text, color=0xFFFF00, x=96, y=11)
        self.splash.append(clock_area)

        # Control State area
        state_text = "State:OFF"
        state_area = label.Label(terminalio.FONT, text=state_text, color=0xFFFF00, x=2, y=25)
        self.splash.append(state_area)
        
        # Wifi reception area
        reception_bitmap = displayio.Bitmap(11, 10, 2)
        reception_palette = displayio.Palette(2)
        reception_palette[0] = 0x000000  # Black
        reception_palette[1] = 0xFFFFFF  # White
        hf.render_reception(reception_bitmap, -90)
        reception_sprite = displayio.TileGrid(reception_bitmap, pixel_shader=reception_palette, x=80, y=6)
        self.splash.append(reception_sprite)

        spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
        esp = adafruit_esp32spi.ESP_SPIcontrol(spi, cb.esp32_cs, cb.esp32_ready, cb.esp32_reset, debug=False)

        #status_light = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2)

        #self.wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(
        #    self.esp, secrets, self.status_light, debug=True
        #)
```

### Behavior

Crashes at "esp =" line (last non-commented line)

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

[bug.zip](https://github.com/adafruit/circuitpython/files/10307030/bug.zip)

dhalbert, 2022-12-27:
Note that other files are here:

> bug.zip

dhalbert, 2023-01-06:
Hi, the problem here is that, at this line
```py
        esp = adafruit_esp32spi.ESP_SPIcontrol(spi, cb.esp32_cs, cb.esp32_ready, cb.esp32_reset, debug=False)
```
you need to pass `DigitalInOut` objects, not `Pin` objects, to the `ESP_SPIcontrol()` constructor. The signature of this constructor is given [here](https://docs.circuitpython.org/projects/esp32spi/en/latest/api.html#adafruit_esp32spi.adafruit_esp32spi.ESP_SPIcontrol) as:
```py
class adafruit_esp32spi.adafruit_esp32spi.ESP_SPIcontrol(spi, cs_dio, ready_dio, reset_dio, gpio0_dio=None, *, debug=False)
```
Note the `dio`, which is short for `DigitalInOut`. The documentation here could be better, and we haven't yet added type annotations to this library, which would also have helped.

Here is a sample program fragment (from [here](https://learn.adafruit.com/adafruit-airlift-breakout/circuitpython-wifi#circuitpython-usage-3085675)):
```py
...
esp32_cs = DigitalInOut(board.D10)
esp32_ready = DigitalInOut(board.D9)
esp32_reset = DigitalInOut(board.D7)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
...
```

OK, you might ask, why did that cause a crash? It's because in CircuitPython 7.3.3 and earlier, the native (C) implementation of `adafruit_busdevice` did not type-check its arguments properly. I reproduced your crash with 7.3.3. However, in 8.0.0-beta.6, I get an exception instead, because we fixed the type-checking:
```
Traceback (most recent call last):
  File "code.py", line 5, in <module>
  File "controllerboardv2.py", line 78, in __init__
  File "adafruit_esp32spi/adafruit_esp32spi.py", line 154, in __init__
TypeError: chip_select must be of type DigitalInOut
```

closed 2023-01-06 by dhalbert

markchapman3216, 2023-01-06:
Thankyou!