adafruit/circuitpython #7388: msgpack doesn't support full range of int32

thetazero, 2022-12-27:
### CircuitPython version

```python
7.3.0-beta.1-4-g7be66a573-dirtyAdafruit CircuitPython 7.3.0-beta.1-4-g7be66a573-dirty on 2022-04-15; PyCubedMini with samd51j19
```

### Code/REPL

```python
>>> msgpack.pack([1672099392], BytesIO())
```

### Behavior

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: no default packer

### Description

This behavior is seen when passing ints greater than 2^30, which seems very strange.

I suspect it has something to do with the call to the function `mp_obj_is_small_int`: [here](https://github.com/adafruit/circuitpython/blob/884371fea7eabdb713a8b4b4f049e0b95efa0a80/shared-module/msgpack/__init__.c#L268). 

For some reason it is not entering
```c
if (mp_obj_is_small_int(obj)) {
        // int
        int32_t x = MP_OBJ_SMALL_INT_VALUE(obj);
        pack_int(s, x);
    }
```
And continues until the failure case
```c
            mp_raise_ValueError(translate("no default packer"));
```

dhalbert, 2022-12-30:
I think this intersects with #6851.

tannewt, 2023-01-04:
@thetazero Small ints are stored in 30 bits themselves because they are stored in pointers with 2 bits to indicate what type of data it is. So, you are right that the small int check isn't correct. It should be more generic to handle any int. (Ints >2**30 are stored as objects rather than small ints in a pointer.)