adafruit/circuitpython #7391: select.poll fails with errno 22 for busio.UART

dennis97519, 2022-12-28:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather RP2040 with rp2040
```

### Code/REPL

```python
# Write your code here :-)
import board
import busio
import sys
# import asyncio
import select

uart = busio.UART(board.TX, board.RX, baudrate=31250)
p = select.poll()
p.register(uart, select.POLLIN)
p.register(sys.stdin, select.POLLIN)

while True:
    res = p.poll()
```

### Behavior

```
Traceback (most recent call last):
  File "code.py", line 14, in <module>
OSError: [Errno 22] Invalid argument
```

### Additional information

Removing `p.register(uart, select.POLLIN)` stops the error from happening

Also happens with asyncio: if I do `reader = asyncio.StreamReader(uart)` and some coroutine does `await reader.read(1)`, asyncio.run() fails with the same OSError with a backtrace pointing at https://github.com/adafruit/Adafruit_CircuitPython_asyncio/blob/main/asyncio/core.py#L176

jepler, 2022-12-28:
I believe this will be fixed in 8, give one of the betas a try.

dhalbert, 2023-01-09:
@dennis97519 Were you able to retry with an 8.0.0 beta?

dhalbert, 2023-10-25:
Closing due to inactivity. Please reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert