adafruit/circuitpython #7393: RuntimeError: pystack exhausted after deep function nesting (WIZnet library)

anecdata, 2022-12-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit QT Py ESP32S2 with ESP32S2
```

### Code/REPL

```python
Placeholder issue: no minimal example at this time. Only happens on some URLs.
```

### Behavior

```
# using latest release adafruit_wiznet5k v1.12.18
Traceback (most recent call last):
  File "code.py", line 111, in get_http
  File "adafruit_requests.py", line 823, in get
  File "adafruit_requests.py", line 662, in request
  File "adafruit_requests.py", line 498, in _get_socket
  File "adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 107, in getaddrinfo
  File "adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 118, in gethostbyname
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 272, in get_host_by_name
  File "adafruit_wiznet5k/adafruit_wiznet5k_dns.py", line 101, in gethostbyname
  File "adafruit_wiznet5k/adafruit_wiznet5k_dns.py", line 122, in _parse_dns_response
  File "adafruit_wiznet5k/adafruit_wiznet5k_socket.py", line 581, in available
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 640, in socket_available
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 889, in socket_read
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 1056, in _get_rx_rcv_size
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 1097, in _read_snrx_rsr
  File "adafruit_wiznet5k/adafruit_wiznet5k.py", line 1156, in _read_socket
RuntimeError: pystack exhausted
```

dhalbert, 2022-12-30:
Might fix for 8.0.0 by increasing PYSTACK or might defer to later. See #7396 and #3362.

BiffoBear, 2022-12-31:
Might this be solved by refactoring Wiznet5k? At the moment, `wiznet5k_socket.gethostbyname` calls `wiznet5k.get_host_by_name` which instantiates an instance of `wiznet5k-dns.DNS` and calls `DNS.gethostbyname` which instantiates a `wiznet5k_socket.socket` and makes calls to `socket.send` and `socket.recv` which calls `WIZNET5k.socket_write`.
This could be refactored into `wiznet5k_socket.gethostbyname` with a couple of functions to build and parse DNS messages and calls to the `WIZNET5K instance that is already linked to the sockets module.. The Wiznet5k DNS module could be removed completely.

anecdata, 2023-04-26:
Closing since e have pystack controls now, and WIZnet library has changed significantly. If this crops up again, we can re-open in the library.