adafruit/circuitpython #7397: error when starting server on pi cow

vithp, 2022-12-29:
CircuitPython Version: Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Raspberry Pi Pico W with rp2040
code.py
`import os
import wifi

ap_ssid = "picow"
ap_password = "picow123"

print()
print("Creating a new Access Point")

#  create access point
wifi.radio.start_ap(ap_ssid, ap_password)
`
output
`
>>> from code import *

Creating a new Access Point
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "code.py", line 15, in <module>
RuntimeError: Wifi is in station mode.
>>>
`
how do i change to ap mode or at least out of station mode?

bill88t, 2022-12-29:
The current implementation of pi cow wifi cannot do both AP and Station at the same time.
To disconnect from the station you are connected from, run `wifi.radio.stop_station()`.

As a reminder, once pi cow AP is started, it cannot be stopped without `microcontroller.reset()` or a powercycle.

vithp, 2022-12-29:
ahh so you cannot stop it via `wifi.radio.stop_ap()` it should be fine with me as i am using it for a webserver

closed 2022-12-29 by vithp