adafruit/circuitpython #7400: GCC 12.2 error on `compile.c:1095`

bill88t, 2022-12-30:
It is a single warning. If it is fixed, circuitpython can be built with latest gcc.
```
../../py/compile.c: In function 'do_import_name':
../../py/compile.c:1095:27: error: 'q_ptr' may be used uninitialized [-Werror=maybe-uninitialized]
 1095 |             qstr q_full = qstr_from_strn(q_ptr, len);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../py/obj.h:33,
                 from ../../py/parse.h:32,
                 from ../../py/scope.h:29,
                 from ../../py/compile.c:33:
../../py/qstr.h:84:6: note: by argument 1 of type 'const char *' to 'qstr_from_strn' declared here
   84 | qstr qstr_from_strn(const char *str, size_t len);
      |      ^~~~~~~~~~~~~~
```
I will try to resolve it by myself.

bill88t, 2022-12-31:
Turns out it only occurs on stackless builds.

dhalbert, 2023-10-28:
Closing as we have moved to gcc 12. Can be reopened if this is still happening.

closed 2023-10-28 by dhalbert