adafruit/circuitpython #7405: After 15 min or so my FeatherS3 crashes unexpectedly

xav-e, 2023-01-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; FeatherS3 with ESP32S3
```

### Code/REPL

```python
# ------------------------------------------------------------------------------------
# First Module
"""
BMP280 -> Sensor Data> Temperature, pressure, and altitude
"""
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""Simpletest Example that shows how to get temperature,
   pressure, and altitude readings from a BMP280"""
# ------------------------------------------------------------------------------------
# Second Module
"""
OLED-FEATHERWING
"""
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: Unlicense
"""
Author: Mark Roberts (mdroberts1243) from Adafruit code
This test will initialize the display using displayio and draw a solid white
background, a smaller black rectangle, miscellaneous stuff and some white text.
"""
# ------------------------------------------------------------------------------------
# Thrid Module
"""DHT11 - Temperature and Humidity measurement"""
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
# SPDX-License-Identifier: MIT
# ------------------------------------------------------------------------------------
# Fourth Module
"""Potenciometers"""
# SPDX-FileCopyrightText: 2018 Anne Barela for Adafruit Industries
#
# SPDX-License-Identifier: MIT
# ------------------------------------------------------------------------------------
# Fifth Module
""" Lux Sensor"""
# SPDX-FileCopyrightText: 2020 Bryan Siepert, written for Adafruit Industries

# SPDX-License-Identifier: Unlicense
# ------------------------------------------------------------------------------------
# Sixth Module
""" DS1307 - RTC Module"""
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT
"""
#Simple demo of reading and writing the time for the DS1307 real-time clock.
# Change the if False to if True below to set the time, otherwise it will just
# print the current date and time every second.  Notice also comments to adjust
# for working with hardware vs. software I2C.
"""
# ------------------------------------------------------------------------------------

# Seventh Module
"""MQTT Broker & Wi-Fi Connection module"""
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT
# ------------------------------------------------------------------------------------

""" Modifications to the models issued by Xavier Zambrano for Aqua-S3"""
# ------------------------------------------------------------------------------------

[... 381 lines cut ...]
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.3 on 2022-08-29; FeatherS3 with ESP32S3
[FeatherS3DriveContents.zip](https://github.com/adafruit/circuitpython/files/10329377/FeatherS3DriveContents.zip)

### Description

The board is the FeatherS3 by Unexpected Maker, it runs okay for like 15min or so and everything works in the server side too, but then it crashes and I do not know why. I've tried to increase the delays, commented out the unsubscribe and disconnect blocks and moved out the connect and subscribe blocks outside the infinite loop. Prior to this changes, all these blocks were run sequentially, but it still crashed after a while.

I admit I'm still getting the hang of all this, but I honestly see no error in neither the wiring nor the code. The only thing I can think of its maybe one of the sensors crashes after a while so everything then crashes.

Or maybe is my internet connection, I've seen that when my UPS internal relays switch to stabilize the power grid voltage of my house this happens too, but as of writing this has happened without that happening

### Additional information

The board is the FeatherS3 by Unexpected Maker, it runs okay for like 15min or so and everything works in the server side too, but then it crashes and I do not know why. I've tried to increase the delays, commented out the unsubscribe and disconnect blocks and moved out the connect and subscribe blocks outside the infinite loop. Prior to this changes, all these blocks were run sequentially, but it still crashed after a while.

I admit I'm still getting the hang of all this, but I honestly see no error in neither the wiring nor the code. The only thing I can think of its maybe one of the sensors crashes after a while so everything then crashes.

Or maybe is my internet connection, I've seen that when my UPS internal relays switch to stabilize the power grid voltage of my

dhalbert, 2023-01-03:
Could you try CircuitPython 8.0.0-beta.6 and see if the problem is still present?

jepler, 2023-01-03:
This is superficially similar to #6791 which also has an S3 and a BME280 sensor.

xav-e, 2023-01-03:
> Could you try CircuitPython 8.0.0-beta.6 and see if the problem is still present?

Of course! I'll tried it out and update this on a moment

Thanks for your help

tannewt, 2023-01-17:
@xav-e Is it still happening?

jposada202020, 2023-01-26:
Hello is happening for me in my `Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3` I am trying to connect to my home assistant, as you can see in the following code, I try to restart the micro-controller, however it does not restart, and the if the USB is connected it will freeze my ubuntu machine. I get a 
```
Error:
 [Errno 128] ENOTCONN
```

```python
try:
    while True:
        x, y, z = [
            value / adafruit_lis3dh.STANDARD_GRAVITY for value in lis3dh.acceleration
        ]
        movement = abs(y * 10)
        output = {
            "movx": movement,
        }
        print("Publishing to %s" % mqtt_topic)
        print(movement)
        client.publish(mqtt_topic, json.dumps(output))
        time.sleep(300)

except Exception as e:
    print("Error:\n", str(e))
    print("Resetting microcontroller in 10 seconds")
    time.sleep(10)
    microcontroller.reset()

```

DJDevon3, 2023-04-28:
Eligible for a retest with 8.0.5 stable. I'm at 2 hours stable on the UM FeatherS3 currently.

jepler, 2023-06-21:
Please re-test once we've merged IDF 5 support into CircuitPython 9.

sveinungkb, 2023-08-15:
> Please re-test once we've merged IDF 5 support into CircuitPython 9.

Is there a build where this can be tested now?

tannewt, 2023-08-15:
Not quite. I'm working on it now.

dhalbert, 2023-12-01:
Could you now test with a 9.0.0-alpha release? Thanks.

tannewt, 2024-01-31:
Please comment if this is still an issue and we'll reopen. (Or file a new issue.)

closed 2024-01-31 by tannewt