adafruit/circuitpython #7406: Exception in alarm.time when monotonic_timeis in the past.

Tim8Trudeau, 2023-01-01:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit Feather RP2040 with rp2040

In my experience the minimum alarm period seems to be 1 second. The code below should produce 10 x 100mS sleep periods for a total of 1 second. I need to sample user buttons 10 times a second but also want to make use of light-sleep to reduce power. 
Output:
mtime=8519.37
wake =8520.26
mtime=8520.27
wake =8521.26
mtime=8521.26
wake =8522.26   alarm returns after 1 second, not 0.1 seconds

I changed line 4 to `sleep_alarm = alarm.time.TimeAlarm(monotonic_time=0.0)` to find the minimum alarm period.
This line produces a value exception because monotonic_time is in the past.
```

### Code/REPL

```python
"""Original code to produce 10 alarms with delays of 0.1 second."""
            for loop in range(10):
                mtime = monotonic()
                print(f'mtime={mtime}')
                sleep_alarm = alarm.time.TimeAlarm(monotonic_time=mtime+0.1)
                alarm.light_sleep_until_alarms(sleep_alarm)
                print(f'wake ={monotonic()}')
                sample_buttons()
```

### Behavior

``` python

sleep_alarm = alarm.time.TimeAlarm(monotonic_time=0.0)

Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
  File "litekit.py", line 148, in litekit_main
  File "litekit.py", line 90, in run
ValueError: Time is in the past.
```

### Description

According to the Docs it should be possible to set monotonic_time to a value in the past. The alarm should simply trigger immediately.
From the Docs for alarm.time.TimeAlarm() - `If the given time is in the past when sleep occurs, the alarm will be triggered immediately.`

### Additional information

Is there a way to get `alarm.time.TimeAlarm()` to alarm on intervals of less than 1 second?

jepler, 2023-01-02:
I did a litltle research to see what's happening in the core C code.

In the rp2040 microcontroller, we rely on the pico-sdk routine `rtc_set_alarm` to set an alarm. It has a granularity of 1 second, because it uses a `datetime_t` which has no sub-second resolution. If you want more info about this routine: https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__rtc.html#gad0018e9069fee8d65c3a01f7bd7ae2f4

If you need shorter sleeps, consider just using time.sleep.

jepler, 2023-01-02:
As for the exception, I think it's operating as intended but the documentation could be better.

*constructing a time alarm for a time **already** in the past* gets you an exception when you create it.

constructing a time alarm for a time in the future but *actually sleeping on that alarm later, after the deadline is past* is supposed to return immediately if its deadline is in the past.

For instance, if you
```python
 time_alarm = TimeAlarm(monotonic_time = time.monotonic() + 1)
 .... do 5 seconds of processing
 alarm.light_sleep_until_alarms(time_alarm)
```it's intended that the light sleep return immediately.

Faced with your questions I'm not adamant that this is the best design, I'm just explaining what I understand of the core code as it's written now. Do you have a use case for creating a TimeAlarm in the past, or is the behavior something you encountered while trying to understand the "always sleeps 1 second" behavior?

PR #7417 (2023-01-05): Improve time alarm doc
Fixes #7406
commit 6b3ae630 (2023-01-05): Improve time alarm doc

Fixes #7406
closed 2023-01-05 by tannewt