adafruit/circuitpython #7411: Hard crash using Circuitpython on Feather SPI32 S3

will49mac, 2023-01-03:
My code will actually  execute and complete, but ends in safe mode. Shorter other bpm280 code piece will run correctly. Control +C  has been unresponsive except in REPL.
Code top and REPEL as jpeg
[image]
text version of code[Hard crash code from ciruit.py

# while loop installed; no  rate code rate call in time wait loop
# Cambria  8-31-2022 8 56 PM text backup, 1-2-2023 version 1 second wait
# Water Tanks 7 26 2022 working needs mode added and valve improved, cleanup
# SPDX-FileCopyrightText: 2018 Kattni Rembor for Adafruit Industries
#
# SPDX-License-Identifier: MIT
"""CircuitPython Essentials Internal RGB LED red, green, blue example"""
import time
# import datetime
import board
import digitalio
import neopixel
import adafruit_mprls
import ssl
import adafruit_bmp280
#  from random import randint
import adafruit_requests
import socketpool
import wifi
from adafruit_io.adafruit_io import IO_HTTP, AdafruitIO_RequestError
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT
# from digitalio import digitalio.DigitalInOut
# set up switch input boad led on
sw_in = digitalio.DigitalInOut(board.IO7)
sw_in.switch_to_input(pull=digitalio.Pull.UP)
print("switch value io7", sw_in.value)
print("start of code water tank")
# set up neopixel
neo = neopixel.NeoPixel(board.NEOPIXEL, 1)
# default run rate to one hour
# Add rate check in end of code count down
defaultrate = 0
rate = defaultrate
print("rate is ", rate)
#  repeating loop start
while True:
    print("Water Tank Level Code Running")
    print("iod7 to gound for sw_status False")
    print("switch value io7", sw_in.value)
    # Turn on neo white for pressure measuremnets
    neo.brightness = 0.2
    neo[0] = (255, 255, 255)
    #
    #  CircuitPython Essentials Internal RGB LED red, green, blue example
    # led = digitalio.DigitalInOut(board.LED)
    # led.direction = digitalio.Direction.OUTPUT
    # from digitalio import digitalio.DigitalInOut
    j = 0
    # while j < 5:
    #    led.value = True
    #    time.sleep(0.5)
    #    led.value = False
    #    time.sleep(0.5)
    #    j = j + 1
    led.value = False
    # set up switch inpu boad led on
    # sw_in = digitalio.DigitalInOut(board.IO7)
    # sw_in.switch_to_input(pull=digitalio.Pull.UP)
    # set up neopixel
    # neo = neopixel.NeoPixel(board.NEOPIXEL, 1)
    # turn it off with 0 value
    print("RGB at 0.2 brightness")
    neo.brightness = 0.2
    # red neo RGB indicates start of measurements
[... 183 lines cut ...]
    #    If not zero, decrement total time by one second
    while total_seconds > 0:
        neo.brightness = 0.2
        neo[0] = (0 , 255, 0)
        try:
            # Get the 'rate' feed from Adafruit IO
            rate_feed = io.get_feed("rate")
        except AdafruitIO_RequestError:
            #   If no 'rate' feed exists, create one
            rate_feed = io.create_new_feed("rate")
        # test of rate
        # print("IO rate value =", rate)
        #  Delays the program one second
        time.sleep(1)
        # Reduces total time by one second
        total_seconds -= 1
        print("countdown at ", total_seconds)
        # add code to check if rate was changed
    print("Bzzzt! The countdown is at zero seconds!")
    neo.brightness = 0.2
    neo[0] = (150, 150, 0)
    time.sleep(3)
    neo.brightness = 0.2
    neo[0] = (0 , 255, 0)
  #   ]

dhalbert, 2023-01-04:
Could you test with 8.0.0-beta.6 from here: https://circuitpython.org/board/unexpectedmaker_feathers3/ ? Thanks. It could easily still not be fixed, but worth trying the latest. (We know of I2C problems on S3.)

will49mac, 2023-01-05:
I loaded uf2 circuit python from unexpected maker 8.0-beta.6, and used the adafruit  8.0 bundle . The unexpected maker had .bin version I was not confident in loading .bin.
I have not had a safe mode happen since. I attached screen shot of where code did fail on eight pass, but no safe mode was invoked. it failed on internet IO call. io.get.rate_feed It appears to me the 8.0.0-beta.6 solves my issue.Thanks you for the suggestion!!!

tannewt, 2023-01-05:
Thanks for the update @will49mac

closed 2023-01-05 by tannewt