adafruit/circuitpython #7419: How to configure GPIO in pulseio.PulseIn

musimasami, 2023-01-05:
When I use pulseio.PulseIn , how to configure GPIO (internal pull-up or pull-down,or set to high impedance on ESP32 C3)

tannewt, 2023-01-05:
PulseIn doesn't support internal pulls because the pin may be connected to a peripheral that isn't gpio. We also assume any needed pulls are external. It should be high impedance unless a trigger duration is provided to resume. It'll push the values for the trigger pulse.

dhalbert, 2023-01-09:
@musimasami Does this answer your question?

dhalbert, 2023-01-26:
I'm a little confused. 

The LED is on `board.D1`, is that right?

Does `PulseIn` work the first time, but after `deinit()`-ing the `PulseOut`, `PulseIn` no logner works? Is that what you are saying is not working?

musimasami, 2023-01-27:
I think PulseIn may not works .

musimasami, 2023-01-28:
Board (Can detect Ir remote signal)
  M5Atom lite,ESP32 s2 . pca10059 , raspberry pi pico , RP2040-Zero
Board(Can't detect ir remote signal)
  ESP32 C3 ( Seeed Studio XIAO ESP32C3 , M5Stack Stamp C3)

DavePutz, 2023-01-28:
I have verified that the call to xRingbufferReceive() in update_internal_buffer() works on an ESP32S3, but always returns 0 on an ESP32C3. So, no pulses will ever show up. I'll continue to investigate.

rysin-user, 2024-07-07:
Jest problem z PulseIn w RP2040. Zamiast zwracać długości impulsów zwraca nazwę obiektu. 

pulsein = pulseio.PulseIn(board.GP22, maxlen=120, idle_state=True)

print(pulsein) 
<PulseIn>

print(pulsein[2]) 
532

print(len(pulsein)) 
72 

Co może być przyczyną?

dhalbert, 2024-07-07:
(Please ask for support on https://adafru.it/discord or https://forums.adafruit.com/viewforum.php?f=60. Don't piggyback on a closed issue.)

> Jest problem z PulseIn w RP2040. Zamiast zwracać długości impulsów zwraca nazwę obiektu.
> [English] There is a problem with PulseIn in RP2040. Instead of returning the pulse lengths, it returns the name of the object.

You have to use the subscripts to get the values. The object itself is not a sequence, but it acts like a sequence in other contexts. As noted here, https://docs.circuitpython.org/en/latest/shared-bindings/pulseio/index.html#pulseio.PulseIn, the object has `__len__()` and `__getitem__()`. it would probably be nice if it were iterable, but it isn't.