adafruit/circuitpython #7427: Adafruit CircuitPython 8.0.0-beta.6-25-gbb3a1c0a2 on 2023-01-04; Adafruit PyPortal with samd51j20

anecdata, 2023-01-06:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6-25-gbb3a1c0a2 on 2023-01-04; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
# Exception
```

### Behavior

I'm able to reproduce this. With the following sequence of events, an Exception printed in the terminal title bar persists after a restart:

```Traceback (most recent call last):
  File "code.py", line 1159, in <module>
IndexError: index out of range

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```
• Re-save code to fix the exception
• hard reset the device (press the button)
```
[tio 21:45:05] Disconnected
[tio 21:45:12] Connected
```
Code restarts (`microcontroller.ResetReason.UNKNOWN` / `supervisor.RunReason.STARTUP`), but the title bar doesn't change:
[image]

tannewt, 2023-01-17:
I'm not able to reproduce this on pico w or samd21 qtpy with tio on Linux.

The status bar is updated before and after code.py runs. It is also updated by request when USB CDC is connected:

https://github.com/adafruit/circuitpython/blob/79b76f7bea6fae11f458ea1ccba0f8649d746d10/supervisor/shared/usb/usb.c#L284

anecdata, 2023-01-19:
I wonder if it's the `tio` connection delay.

tannewt, 2023-01-20:
> I wonder if it's the `tio` connection delay.

I'm using tio too so I don't think its that.

tannewt, 2023-04-14:
@anecdata are you still seeing this?

anecdata, 2023-04-17:
Still occurring with `Adafruit CircuitPython 8.1.0-beta.1-25-g4d7e54310 on 2023-04-16; Adafruit PyPortal with samd51j20`.

Even simpler to demo is to control-C during run, the `KeyboardInterrupt` pops into the title bar. Hard reset, and the `KeyboardInterrupt` is still in the title bar after the `tio` reconnects and `code.py` is runing and printing to the serial console.

The title bar otherwise stays current: for example, control-C then control-D will refresh away the `KeyboardInterrupt`.

(Environment: macOS 13.3.1, Terminal.app, tio)

dhalbert, 2023-04-17:
The title bar is not sent if USB is not connected, so, yes, this sounds like a timing issue.