adafruit/circuitpython #7431: Incorrect behaviour of timeout on socket / socket & serial port blocked - CP 8.0.0 beta6

yousafs, 2023-01-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32-S3
Also tested and replicated on ESP32-S2 board
```

### Code/REPL

```python
from adafruit_httpserver.mime_type import MIMEType
from adafruit_httpserver.request import HTTPRequest
from adafruit_httpserver.response import HTTPResponse
from adafruit_httpserver.server import HTTPServer

pool = socketpool.SocketPool(wifi.radio)
server = HTTPServer(pool)
server.request_buffer_size = 2048
server.socket_timeout = 1
server.start(str(wifi.radio.ipv4_address))

@server.route("/")
def base(request: HTTPRequest):
    with HTTPResponse(request, content_type=MIMEType.TYPE_TXT, chunked=False) as response:
        try:
            print("REQUEST: /")
            response.send("HELLO")
        except OSError:
            response.send("ERROR")

while True:
    try:
        led.value = True
        time.sleep(0.05)
        led.value = False
        time.sleep(0.1)
        server.poll()
        
    except OSError as error:
        print(error)
        continue
```

### Behavior

If you use the code attached on cp 8.0.0 and make a request to the server from Chrome (which makes the request you ask for and then immediately a second request to keep a port open), the loop will stop for 60 seconds until Chrome closes the open socket. I would expect setting `server.socket_timeout` to 1 to abort the connection but it just stays blocked, seemingly at `sock.recv_into()` in the server.py include of the HTTP Server project.

### Description

This issue was found whilst investigating a problem with Chrome accessing the Adafruit HTTP Server together with @michalpokusa, original issue is here:

[https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/31]( #https://github.com/adafruit/Adafruit_CircuitPython_HTTPServer/issues/31)

If a socket is opened and nothing is sent down it (as Chrome will do after every request), the socket will stay open and block, the `server.socket_timeout` setting does not seem to have any effect.

@michalpokusa also observed that while the code is stuck on `sock.recv_into()` the serial port is also blocked.

### Additional information

- This problem does not appear on CircuitPython 7.3.3, only on 8.0.0
- Problem exists on both ESP32-S2 and ESP32-S3
- On CP 7.3.3 after processing request program loops .pool(), on CP 8.0.0 most of the time it gets stuck on sock.recv_into()
- Serial connection seems to be blocked when stuck on sock.recv_into()
- The problem could be related to connections which are opened and no data sent down them

dhalbert, 2023-01-12:
I have reproduced this problem with Chrome, with the minute-long stall (didn't measure exactly, but it's long). Firefox, stalls too, but the stall is only 5 seconds. In both cases, it takes a few fetches to cause a stall, as if it's using up available sockets. Repeating a `curl` fetch every second causes no problems.

dhalbert, 2023-01-13:
All of a sudden I can no longer reproduce this problem. I'm not sure why at all, but Chrome recently got upgrade to `Version 109.0.5414.74 `.

I am still seeing an odd thing where at the beginning the `poll()` is being called constantly but after one or more fetches, it is being called only once after each fetch. I am suspicious there is still an issue.

I tried HTTPServer 2.1.0 and 2.2.0 to make sure there was not an accidental fix on the HTTPServer side.

closed 2023-01-16 by MicroDev1

kakashi277, 2023-02-26:
**I have this problem too, w5500-evb-pico use http_server  , not response after 1 minute not have income request.**   
CircuitPython Bundle 8.0.3

michalpokusa, 2023-02-26:
> **I have this problem too, w5500-evb-pico use http_server , not response after 1 minute not have income request.** CircuitPython Bundle 8.0.3

Can you provide some code example? How are you accessing your HTTP server? curl? Chrome? Firefox? This error was preety much only visible when using Chrome.

I want to check whether we are talking about precisely the same error.

Sinai, 2023-07-24:
Hi,

I have the same issue.

CircuitPython 8.2.0 , adafruit_wiznet5k 3.0.0 ,

cs = digitalio.DigitalInOut(board.D10)
spi_bus = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
eth = WIZNET5K(spi_bus, cs)
socket.set_interface(eth)
socket.setdefaulttimeout(0.0) ------>>>>>> timeout
server = Server(socket, "/test")
while True:
  try:
     server.poll()
  expect...........
  
  
server.poll() never returns even if i have defined a timeout for the socket.
I am using chrome  115.0.5790.102

Thank you

PR #7455 (2023-01-16): espressif: always make new client sockets be non-blocking
Fixes #7431.

- The new client socket returned by `accept()` was not set to `O_NON_BLOCK` all the time, causing it to block until a request was made. Now ensure that it is always set to non-blocking.

Other fixes:

- `write()`s to `eventfd`s should always be 8 bytes. There was a write that was only 4 bytes (writing the `fd`). This should actually return an error, but it wasn't checked. @tannewt I don't see that the results of these writes are used anywhere, so I think this change is OK, but you can take a look. Internally they are added [... cut]