adafruit/circuitpython #7433: TinyPico v1 flash read err 1000

luigi311, 2023-01-10:
### CircuitPython version

```python
Adafruit Circuitpython v8.0.0-beta.6;
Adafruit Circuitpython 20230106-037bfb3;
```

### Code/REPL

```python
N/A
```

### Behavior

```bash
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
```

### Description

Clean tinypico with
esptool.exe --chip esp32 --port COM9 erase_flash

then flash circuitpython via
esptool.exe --chip esp32 --port COM9 write_flash -z 0x1000 ..\adafruit-circuitpython-unexpectedmaker_tinypico-en_US-8.0.0-beta.6.bin

No drive shows up and connecting via putty outputs the error messages

### Additional information

This is on the original run of tinypicos so the v1 which is confirmed by the esptool when flashing. I am able to install and run micropython.

dhalbert, 2023-01-10:
My understanding is that `esptool.exe` is obsolete. We use `esptool.py`. I am not sure what `esptool.exe`s command line args are but, we flash CircuitPython starting at 0x0000, not 0x1000. Could you try that?

luigi311, 2023-01-10:
looking at the output for esptool.exe it is just a wrapper for the python script for windows 

 .\esptool.exe --chip esp32 --port COM9 write_flash -z 0x1000 ..\tinypico-20220618-v1.19.1.bin
esptool.py v4.4

Just in case though i did install esptool from pip and ran that with the 0x0000 and that seems to be working so it might of just been the 0x1000 vs 0x0000 difference. 

I am able to connect via putty serial  and hello world completes but i do not see the circuitpy drive, according to the cricuitpython getting started that should appear once connected

dhalbert, 2023-01-10:
> I am able to connect via putty serial and hello world completes but i do not see the circuitpy drive, according to the cricuitpython getting started that should appear once connected

The TinyPico is a plain ESP32, which does not have native USB, so it cannot present a CIRCUITPY drive. You'll have to use Thonny or similar to load files onto it. The REPL is via the USB-serial converter cip.

luigi311, 2023-01-10:
I see, thanks for all the info. Ill go ahead and look into what i can use for this and use Thonny to start my search. Looks like you solved all my issues so i can go ahead and close the ticket, thanks for the speedy help

closed 2023-01-10 by luigi311

dhalbert, 2023-01-10:
Glad things are working. Check out https://learn.adafruit.com/circuitpython-with-esp32-quick-start for help with such boards. It does not mention Thonny (but maybe it should). It talks about using the web workflow which is new and a bit experimental.

sctv, 2023-01-10:
https://learn.adafruit.com/circuitpython-with-esp32-quick-start/setting-up-web-workflow#option-2-create-settings-dot-toml-file-using-thonny-3124379 is the sublink.