adafruit/circuitpython #7442: bootloader problems when porting new ESP32S3 board to CP

hamishcunningham, 2023-01-11:
We've been porting a new board to CP following these instructions: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

We rebuilt and flashed CP for an existing board (the ESP32S3 Feather with 4mb flash 2mb psram) and verified that it worked fine, but now that we're porting to our own board (that uses ESP32S3 8mb flash / 8mb psram) we're seeing a bootloop early in the boot process, e.g.:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378054
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x68
load:0x403c9700,len:0x980
load:0x403cc700,len:0x25a0
entry 0x403c98a4
ESP-ROM:esp32s3-20210327
... (loops) ...
```

We also tried porting tinyuf2 from https://github.com/adafruit/tinyuf2 but this time we don't get as far as presenting as a USB device, e.g.:

```
[Jan 9 11:57] usb 1-2.1.4: USB disconnect, device number 35
[ +0.649994] usb 1-2.1.4: new full-speed USB device number 36 using xhci_hcd
[ +0.288069] usb 1-2.1.4: device descriptor read/64, error -32
[ +0.396033] usb 1-2.1.4: device descriptor read/64, error -32
[ +0.415971] usb 1-2.1.4: new full-speed USB device number 37 using xhci_hcd
[ +0.568096] usb 1-2.1.4: device descriptor read/64, error -32
[ +0.415989] usb 1-2.1.4: device descriptor read/64, error -32
[ +0.108353] usb 1-2.1-port4: attempt power cycle
[ +0.607433] usb 1-2.1.4: new full-speed USB device number 38 using xhci_hcd
... (repeats) ...
```

We've tried to verify config by copying from other boards with similar setup, and tried various config options, but no luck...

Any ideas about what to try next?

Thanks for all the great docs! A couple of suggestions for https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

- include a link to where a valid UF2 bootloader build would come from (we assume tinyuf2)?
- this paragraph points to code that doesn't contain the config options that it describes: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython?view=all#uf2-bootloader-3033719
- add description of how to prep a board to be included on https://circuitpython.org/board/...?

Thanks!
Hamish
 
PS Our fork is here: https://github.com/hamishcunningham/circuitpython and the diffs to main are here: https://github.com/adafruit/circuitpython/compare/main...hamishcunningham:circuitpython:unphone

PPS Our board is described here:

- docs: https://iot.unphone.net/#unphone-yourself
- schematic: https://gitlab.com/hamishcunningham/unphone/-/blob/master/doc/unphone-spin9-schematic.pdf
- Arduino library: https://gitlab.com/hamishcunningham/unphonelibrary
- https://unphone.net/the-unphone/
- our CP build is in this Dockerfile: https://gitlab.com/hamishcunningham/the-internet-of-things/-/blob/master/support/circuitpython/Dockerfile

MicroDev1, 2023-01-11:
> (that uses ESP32S3 8mb flash / 8mb psram)

Might be similar to #5999. See: https://github.com/adafruit/circuitpython/pull/5999#issuecomment-1320807909

tannewt, 2023-01-11:
Please clean and rebuild CircuitPython with `DEBUG=1` That should enable debug prints from CP and the ESP-IDF's second stage bootloader which should appear after the `ESP-ROM` print. (TinyUF2 runs after the second stage bootloader when present. It isn't needed for CircuitPython to run though.)

hamishcunningham, 2023-01-12:
Thanks, with DEBUG=1 I get:
```
Invalid image block, can't boot.
```

(Not sure if this is relevant, but I only get serial output if I flash `circuitpython-firmware.bin`; nothing if I flash `firmware.bin`.)

hamishcunningham, 2023-01-12:
I got a little further :)

I made a few more [changes to board config](https://github.com/hamishcunningham/circuitpython/commit/bf334005520c5004e986bf5faf6e8a3cee13232e) and can now get CIRCUITPY to appear if I flash `circuitpython-firmware.bin` from my build. The serial output is this though:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

The CIRCUITPY tree doesn't contain very much:
```
$ ls -lR
.:
total 2
-rw-r--r-- 1 hamish hamish   22 Jan  1  2000 code.py
drwxr-xr-x 2 hamish hamish 1024 Jan  1  2000 lib
-rw-r--r-- 1 hamish hamish    0 Jan  1  2000 settings.toml

./lib:
total 0
$ cat code.py 
print("Hello World!")
```

Any ideas? Tnx!

tannewt, 2023-01-12:
Debug builds on ESP should stop in the ESP-IDF panic handler and give you a backtrace. Did you stop making a DEBUG build?

hamishcunningham, 2023-01-12:
With DEBUG it gives me:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3c120020,len:0x4a0a8
load:0x3fc9cdb0,len:0x5edc
load:0x40374000,len:0x64
load:0x42000020,len:0x11d70c
Invalid image block, can't boot.
ets_main.c 329 
```

hamishcunningham, 2023-01-13:
Ah, I seem to have peeled another layer off the onion! This looks more interesting:
```
I (421) esp_image: segment 4: paddr=00173118 vaddr=40378434 size=14150h ( 82256) load
I (440) esp_image: segment 5: paddr=00187270 vaddr=600fe000 size=0002ch (    44) load
I (440) esp_image: segment 6: paddr=001872a4 vaddr=600fe030 size=01010h (  4112) load
I (456) boot: Loaded app from partition at offset 0x10000
I (457) boot: Disabling RNG early entropy source...
I (469) opi psram: vendor id : 0x0d (AP)
I (469) opi psram: dev id    : 0x02 (generation 3)
I (469) opi psram: density   : 0x03 (64 Mbit)
I (473) opi psram: good-die  : 0x01 (Pass)
I (477) opi psram: Latency   : 0x01 (Fixed)
I (482) opi psram: VCC       : 0x01 (3V)
I (487) opi psram: SRF       : 0x01 (Fast Refresh)
I (492) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (498) opi psram: BurstLen  : 0x01 (32 Byte)
I (503) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (509) opi psram: DriveStrength: 0x00 (1/1)
W (514) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (524) spiram: Found 64MBit SPI RAM device
I (529) spiram: SPI RAM mode: sram 80m
I (533) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (540) cpu_start: Pro cpu up.
I (544) cpu_start: Starting app cpu, entry point is 0x4037848c
I (0) cpu_start: App cpu up.
I (976) spiram: SPI SRAM memory test OK
I (986) cpu_start: Pro cpu start user code
I (986) cpu_start: cpu freq: 240000000
I (986) cpu_start: Application information:
I (988) cpu_start: Project name:     circuitpython
I (994) cpu_start: App version:      8.0.0-beta.6-114-gbf33400552-di
I (1001) cpu_start: Compile time:     Jan 13 2023 13:54:15
I (1007) cpu_start: ELF file SHA256:  b90961bc2e2b37ed...
I (1013) cpu_start: ESP-IDF:          8d0f1abce
I (1019) heap_init: Initializing. RAM available for dynamic allocation:
I (1026) heap_init: At 3FCADF88 len 0003B788 (237 KiB): D/IRAM
I (1032) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1039) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1045) heap_init: At 600FF040 len 00000FC0 (3 KiB): RTCRAM
I (1053) spi_flash: detected chip: generic
I (1057) spi_flash: flash io: dio
I (1061) sleep: Configure to isolate all GPIO pins in sleep state
I (1067) sleep: Enable automatic switching of GPIO sleep configuration
I (1075) coexist: coexist rom version e7ae62f
I (1080) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (10) gpio: GPIO[21]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (10) gpio: GPIO[33]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (20) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1030) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1040) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1040) gpio: GPIO[1]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1050) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1060) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1070) gpio: GPIO[4]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1080) gpio: GPIO[5]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1090) gpio: GPIO[6]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1100) gpio: GPIO[7]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1110) gpio: GPIO[8]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1110) gpio: GPIO[9]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1120) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1130) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1140) gpio: GPIO[12]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1150) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1160) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1170) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1180) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1190) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1200) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1210) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1220) gpio: GPIO[38]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1230) gpio: GPIO[45]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1240) gpio: GPIO[46]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1250) gpio: GPIO[47]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1260) gpio: GPIO[48]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
E (1260) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
E (1270) I2S: i2s_driver_uninstall(2047): I2S port 1 has not installed
E (1280) timer_group: timer_deinit(312): HW TIMER NEVER INIT ERROR
E (1290) timer_group: timer_deinit(312): HW TIMER NEVER INIT ERROR
E (1290) timer_group: timer_deinit(312): HW TIMER NEVER INIT ERROR
E (1300) timer_group: timer_deinit(312): HW TIMER NEVER INIT ERROR
```
The I2S error looks like a pin definition error maybe? I should change something in pins.c maybe?

HW TIMER I have no idea tho...?

Tnx!

tannewt, 2023-01-13:
Those errors are normal because we're resetting devices that haven't been used. This looks like it is starting up correctly to me.

hamishcunningham, 2023-01-13:
Must be getting closer :)
I rebuilt without the DEBUG flag and now I get a drive appearing, though it is empty and labelled "6MB Volume".
On serial I'm back to the Hardfault_Handler:

```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.
```

Any ideas? Tnx!

dhalbert, 2023-01-13:
I don't know about the HardFault_Handler safe mode issue, but try doing this to re-create CIRCUITPY:
```py
>>> import storage
>>> storage.erase_filesystem()
```

These errors are innocuous. They show up on every port:
```
E (1260) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
E (1270) I2S: i2s_driver_uninstall(2047): I2S port 1 has not installed
```

hamishcunningham, 2023-01-13:
Thanks, but I don't get a REPL and code.py isn't recognised. The drive that pops up when the device boots contains nothing :(

hamishcunningham, 2023-01-15:
I'm confused about the difference between the bootloader(s) that a CP build creates and [TinyUF2](https://github.com/adafruit/tinyuf2).

Might it help for me to port my board to tinyuf2?

dhalbert, 2023-01-15:
TinyUF2 is a secondary bootloader. It is started by the initial bootloader, and knows to start the "main" application if it is present.

You don't need to do that right now. Get the .bin version working.

Looking at your https://github.com/adafruit/circuitpython/compare/main...hamishcunningham:circuitpython:unphone diffs, the size of flash and PSRAM is still 4MB/2MB, not 8/8, as you say your board is. Also take a look at the 8MB/0MB boards we support. You may need to change the `qio` to `dio` in `mpconfigport.mk` as well. (But that is used only for the flash program args, I believe.)

hamishcunningham, 2023-01-15:
Thanks! But I think I've already made those changes...? E.g.:

[mpconfigboard.mk](https://github.com/hamishcunningham/circuitpython/blob/unphone/ports/espressif/boards/unphone/mpconfigboard.mk#L8)
[sdkconfig](https://github.com/hamishcunningham/circuitpython/blob/unphone/ports/espressif/boards/unphone/sdkconfig)

Our chip is an ESP32S3 N8R8 so I've tried to match settings with the [Espressif N8R8 devkit](https://github.com/hamishcunningham/circuitpython/tree/unphone/ports/espressif/boards/espressif_esp32s3_devkitc_1_n8r8). (I started from a feather definition, but figured that wasn't correct.)

dhalbert, 2023-01-15:
Ah, I was looking at the oldest commit instead of the newest commit, sorry!

dhalbert, 2023-01-15:
Do you have a DevKit N8R8 or N8R2 in hand, to test whether the definitions we have work for those boards? If those work and the board files are very similar to yours, then I'd guess the issue is just what differs between your board and those. But I may be saying the obvious - if so, sorry. I'm wondering about strapping pins, etc.

I have a DevKit N8R2 but not an N8R8. I will add it to my next staff order.

hamishcunningham, 2023-01-15:
Aha, I tweaked some more pindefs and now it works! Whahey!
```Adafruit CircuitPython 8.0.0-beta.6-38-gfd7f36ee69-dirty on 2023-01-15; unPhone with ESP32S3```
THanks for your help, much appreciated :)

Given that I've done a "make all" and a .uf2 has been generated and then (I think) combined with the (first stage?) bootloader.bin, does that mean I'm now running UF2, or is that another step?

Thanks again!

dhalbert, 2023-01-15:
Fantastic!

If you are loading a .bin, then you are not using UF2. You need to load `combined.bin` from a TinyUF2 build onto the board (after erasing it completely). Then TinyUF2 will run after the built-in bootloader. If you do a double-click, it would go into UF2 mode and present a BOOT drive as an MSC USB device, and you would copy the UF2 onto it.

UF2 is a special format that contains information about "put this block of data here", etc.

You may be able to start with https://github.com/adafruit/tinyuf2/tree/master/ports/espressif/boards/espressif_esp32s3_devkitm_1 and modify it as necessary. Then you load up the combined.bin as described, for example, here: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader

hamishcunningham, 2023-01-15:
Grand, tnx! Have a good one. H

closed 2023-01-15 by hamishcunningham

SenelCanik, 2023-02-03:
> Grand, tnx! Have a good one. H

Hi hamishcunningham. We have similar problem. We made our own board with ESP32-S3 (FN8 Model with 8 MB Flash internal).   After looading bootloader.bin (and many different *.bin files) we get the following messages from COM port. We use Windows CMD for esptool.exe. How did you solve the problem?

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40043ac8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1634
load:0x403b6000,len:0xe74
load:0x403ba000,len:0x31c8
Checksum failure. Calculated 0x8e stored 0xde
ets_main.c 329 
... (loops) ...