adafruit/circuitpython #7446: Safe mode on STM32F405

teejay0909, 2023-01-12:
[CIRCUITPY_contents.zip](https://github.com/adafruit/circuitpython/files/10397772/CIRCUITPY_contents.zip)
Was learning how to use numpy on a basic script. Then got an error message requesting I send a copy of my CIRCUITPY drive. I hope this helps community improve. I zipped my entire drive and attached.

dhalbert, 2023-01-12:
Thanks. Could you try the 8.0.0-beta.6 build here: https://circuitpython.org/board/feather_stm32f405_express/ ?

There have been some fixes to ulab (and many other things).

teejay0909, 2023-01-12:
I've cleared/passed the error. I think it was due to initializing an empty
array. As my temporary workaround I pre-allocated an array of zeros.

ulab.numpy.array() #caused error
ulab.numpy.array([]) # caused soft boot from fatal crash.
ulab.numpy.array((0,0,0)) # is my workaround.

Haven't got the time to do beta 8x.

dhalbert, 2023-01-12:
I am not sure what you're trying to do with the `concatenate()`. You have:

`big_array=np.concatenate((sensor.acceleration), axis=1)`

`sensor.acceleration` is a 3-tuple, like `(2.0, 4.0, 6.0)`. `np.concatenate()` takes a sequence of arrays as its first argument, but you passed it a sequence of scalars. In desktop Python, the above causes this error:
```py
>>> np.concatenate((2.0,4.0,6.0), axis=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<__array_function__ internals>", line 5, in concatenate
ValueError: zero-dimensional arrays cannot be concatenated
```
With `ulab`, it crashes, which it should not do, and I'll report that to the `ulab` author.

But perhaps your code is not the above anymore. If you've solved the problem, that's great.

v923z, 2023-01-12:
> I've cleared/passed the error. I think it was due to initializing an empty array. As my temporary workaround I pre-allocated an array of zeros. 

```python
ulab.numpy.array() #caused error 
ulab.numpy.array([]) # caused soft boot from fatal crash. 
ulab.numpy.array((0,0,0)) # is my workaround.
```
@teejay0909 It's not quite clear to me, whether these errors are really errors, or part of the bug report on the `concatenate` function. In any case, if there is anything that bugs you (pun intended), you can always [raise an issue](https://github.com/v923z/micropython-ulab/issues), and I would be happy to come up with a fix. As for `concatenate`, I'll sort that out in one or two days.

dhalbert, 2023-01-12:
> ah I missed that, thanks @dhalbert !

Yes, sorry, for not calling that out explicitly in a comment. It was on this page as a cross-reference but not so obvious.

v923z, 2023-01-15:
@dhalbert This should be fixed by https://github.com/v923z/micropython-ulab/pull/575, which has been merged into master.

dhalbert, 2023-11-07:
Closing as fixed upstream.

closed 2023-11-07 by dhalbert