adafruit/circuitpython #7449: Strange Flash memory read-error on ESP32-S2

overflo23, 2023-01-12:
### CircuitPython version

```python
I can write to the 4MB of flash from begininnig to end without a failure (tested with a small .bin from /dev/urandom and verify_flash)
After powercycle i can read it out correctly.
After erase_flash i can read 4MB of 0xff correctly.
After flashing tinyuf2 bootloader i can correctly read all 4 MB back.

Than i install circuitpython 8.0-beta6 (or any other version from 7.1 up)
than executes and works as expected.

But now i can not read the flash anymore?!

esptool.py  read_flash 0 0x400000 test_dump.bin
-> stalled at 933888 bytes.

By hand i can read up to 937983 which is 4095 bytes away (can this be a pagesize?)

Also i can read parts of the memory back if i use a higher offset behind that "faulty page" but never the whole flash.

when tested on another board the same behaviour arises, but at a different flash page.

This might be a bug with tinyuf2 as well, or a hardware issue i have no idea how to debug this any further.
```

### Code/REPL

```python
nope.
```

### Behavior

...

tannewt, 2023-01-12:
Why do you think this is a CircuitPython bug? It isn't running CircuitPython as far as I can tell. We don't do anything to protect the flash. What chip is this?

dhalbert, 2023-01-12:
I've been investigating this in the forum thread. It's an esptool.py bug, but it's very strange. I'll close this and we'll continue there.

closed 2023-01-12 by dhalbert