adafruit/circuitpython #7454: Espressif microphone support

gpeyton, 2023-01-14:
Currently there appears to be no support for microphones using audiobusio.PDMIn or analogbufio, and analogio sampling rate is too slow. 

Are there any workarounds or plans to support mic input? Using either PDM or analog input.

dhalbert, 2023-01-14:
We have no immediate plans, but it's something we'd like to do long term. `analogbufio` is quite new.

Related issues:
- #6969
- #5456

gpeyton, 2023-01-15:
@dhalbert thanks. Are there any workarounds to be able to utilize any type of microphone with Espressif?

dhalbert, 2023-01-15:
Not right now, sorry, without external hardware. Adding an implementation of `analogbufio` for Espressif is probably the easiest thing to do, `PDMIn` maybe somewhat less so. `I2SIn` would need some API definition. Would you be interested in working on that? We mentor folks who would like to contribute implementations to CircuitPython.

gpeyton, 2023-01-15:
@dhalbert yes, I would be interested in helping if you can provide some direction?

dhalbert, 2023-01-15:
Sure! It looks like you already have had success building CircuitPython, based on your repos. For this particular thing, what you need to do is reimplement what is in `ports/raspberrypi/common-hal/analogbufio/` for `ports/espressif`. The Python API side of the work is already done, in `shared-bindings`, so you won't need to reimplement that.

So you can start by making a branch in your fork, and copying over the `common-hal` sources mentioned above. Then you will replace the insides :slightly_smiling_face: . You'll want to study the Espressif doc for ADC DMA. It may be similar to other DMA tasks, such as SPI (I haven't looked!).

One question is whether the API varies a lot between the different Espressif chips. You may want to choose a subset of which chips to support if that is the case.

You need to set `CIRCUITPY_ANALOGBUFIO = 1` in `ports/espressif/mpconfigport.mk`. That will set up the build system to look for sources for the `analogbufio` implementation when it compiles `espressif` boards

There is some description of all this in https://docs.circuitpython.org/en/latest/docs/common_hal.html, and the other related doc sections (see the left sidebar). There is also https://learn.adafruit.com/extending-circuitpython, but unfortunately it's quite out of date and we haven't had time to update it yet.

We CircuitPython core developers hang out in https://adafru.it/discord in the **#circuitpython-dev** channel, so if you have a quick question you can drop in there. Feel free to start a draft PR, or point us to your branch.

tannewt, 2024-12-04:
#9828 adds pdm support

closed 2024-12-04 by tannewt