adafruit/circuitpython #7460: Serial interface on Adafruit Matrix Portal M4 said to file an issue so I am.

eddtrettel, 2023-01-17:
Here's wht the serial interface said:

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues

Press any key to enter the REPL. Use CTRL-D to reload.

I'll attach all files from the device drive, except for the "secrets.py" file, which I'm not using in this code and wish to keep private anyway.

Note that after the crash the "code,.py" file was empty (zero length), so I'm enclosing the actual code.py file that I was using at the time.

[lib.zip](https://github.com/adafruit/circuitpython/files/10438400/lib.zip)

dhalbert, 2023-01-17:
Thanks, what version of CircuitPython are you using?

Did this happen after an extended period of running, or does it happen reliabily right away (or after a specific series of events)?

eddtrettel, 2023-01-17:
Hi Dan,

Thanks for the quick reply.  I'm a hobbyist and unfamiliar with the process
of reporting an issue, so please bear with me.

I'm running Adafruit CircuitPython 8.0.0-beta.6-28-g6013cde6a on
2023-01-05; Adafruit Matrix Portal M4 with samd51j19

I use the Mu editor on Windows 11 to update and debug my code, as I have
for several years.  The pattern is a predictable one of starting the Mu
editor session, opening the code.py window and the serial window, and then
repeatedly changing the code and doing a Save.

Over the course of one such session I often Save the code.py file dozens of
times, with no issue.

This is the first time ever in my use of the Mu editor that this board
issue has occurred.  It only occurs when I do a Save command, and not at
every Save.

The Mu editor session can be as short as 1-2 minutes and when I save it has
the problem shown.

However, it's not consistent.  It happened today for the first time ever,
but since then I've been able to Save with no issue, other than once or
twice.

When it does this I press the Reset button on the board, which appears to
boot into some funky safe mode, and then I press Reset again and it's
normal.

I'm making edits as normal, albeit with occasional hangs.

BTW, the "USB card" on the board is about 25% full.

Again, I'm a real amateur so please let me know what info you could use.

Thanks,
Edd

eddtrettel, 2023-01-17:
Hi Dan,

I should add that when this fails (doing a Save) the "busy" Windows icon
appears on my screen at the cursor/mouse position.  This spins for about 30
seconds before the serial window shows the text I sent you (crashed, safe
mode, etc).  I had never seen this ever before, and at first assumed that I
should just wait for the Save to finish.

I suppose this could or could not be indicative of a problem with the "USB
drive" (not the proper term, I know) on the Matrix Portal M4?  Failing or
marginal memory on the board?

Again, after this fails the code.py file on the drive has zero bytes.
However, I'm able to salvage my work since the code.py window in the Mu
editor still has the code and I save it to a temporary place elsewhere
until I've reset the Matrix Portal M4 a few times to get it back into
"normal" mode.

Hope this helps.

Thanks again,
E.T.

dhalbert, 2024-03-08:
Sorry for the lack of response on this. It is over a year old. If you're still working on this, please update the bootloader on your Matrix Portal, and try CircuitPython 9.0.0-rc.0 or later. I'll close for now but we can reopen if there is fresh info.

closed 2024-03-08 by dhalbert