adafruit/circuitpython #7472: Hard crash on ESP32S2 using 2 displays

darianbjohnson, 2023-01-21:
CircuitPython 8.0.0-beta.6
Board - Adafruit Feather ESP32s2

Error message:
```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
NLR jump failed. Likely memory corruption.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues
```

Code:
```
import board
import displayio
import vectorio
import terminalio
from adafruit_simplemath import map_range
from adafruit_display_text import bitmap_label, wrap_text_to_lines, label #remove label later
from adafruit_st7789 import ST7789
import adafruit_imageload
from adafruit_bitmap_font import bitmap_font
import time
import gc
import math
import random
import rtc

import hexTileFunctions as hexFuncs

PANTHER_FONT_16 = bitmap_font.load_font("/fonts/Panther-16.bdf")
PANTHER_FONT_25 = bitmap_font.load_font("/fonts/Panther-25.bdf")
PANTHER_FONT_20 = bitmap_font.load_font("/fonts/Panther-20.bdf")
PANTHER_FONT_33 = bitmap_font.load_font("/fonts/Panther-33.bdf")
WEATHER_FONT = bitmap_font.load_font("/fonts/weather-31.bdf")

# Release any resources currently in use for the displays
displayio.release_displays()

# define main display
spi = board.SPI()
tft_cs = board.D9
tft_dc = board.D10

# define menu display
menu_tft_cs = board.D12
menu_tft_dc = board.D11

# set up menu display
menu_display_bus = displayio.FourWire(spi, command=menu_tft_dc, chip_select=menu_tft_cs)
menu_display = ST7789(
    menu_display_bus, rotation=0, width=135, height=240, rowstart=40, colstart=53
)

print(" menu bus", gc.mem_free())

# Make the display context
splash = displayio.Group()
menu_display.show(splash)

menuLabel = bitmap_label.Label(PANTHER_FONT_25, text="1abc\n2def\n3abc\n4def\n5abc\n6def\n7abc\n8def", color=0xFFFFFF, scale = 1,anchor_point = (0.0, 0.0), anchored_position= (0, 0))

splash.append(menuLabel)

# set up maindisplay
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)

display = ST7789(display_bus, width=240, height=320, rotation=0)

print(" display bus 2", gc.mem_free())

while True:
    pass

```
**Background**
I compiled a new version of CPy for the ESP32s2 Feather, and I added a line to the mpconfig.h file to use two displays [#define CIRCUITPY_DISPLAY_LIMIT (2)]

The code compiles; and I am able to get the code to run on a first attempt. However, when I make a code change and save the file, I get the error listed above. The Mu editor looses the ability to write to the file and the board starts to run in safe mode.

I'm not exactly sure what I am doing wrong. Any thoughts?

Note - I am attempting to make a DEBUG version of the code for testing, but I'm running out of room on the board. Any ideas on what files to move so that I can create a DEBUG version?

Breazile, 2023-01-21:
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather ESP32S3 4MB Flash 2MB PSRAM with ESP32S3
Board ID:adafruit_feather_esp32s3_4mbflash_2mbpsram
UID:4F21AFA5124C

```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Please file an issue with the contents of your CIRCUITPY drive at 
https://github.com/adafruit/circuitpython/issues
```

I'm using an external 1.14" display with the ESP32S3 detailed here - https://github.com/Breazile/MandoPuter

It ran for almost 8 hours and then crashed hard. Second time this has happened. I found it crashed overnight and restarted it this morning.

Code is below and contents of the drive is here - https://github.com/Breazile/MandoPuter/blob/master/Releases/ESP32-S3Pre-Beskar.zip
```
"""
MandoPuter will display text in a Mandalorian font on a tiny LCD display

File   - code.py
Author - Jon Breazile

https://github.com/Breazile/MandoPuter

Font credits to ErikStormtrooper, the bitmap fonts were created from his TrueType font
http://www.erikstormtrooper.com/mandalorian.htm
"""
import gc
import time
import alarm
import board
import busio
import pwmio
import neopixel
import digitalio
import displayio
import adafruit_dotstar as dotstar
import adafruit_imageload
from analogio import AnalogIn
from adafruit_display_text import label
from adafruit_bitmap_font import bitmap_font
from adafruit_st7789 import ST7789
from adafruit_lc709203f import LC709203F
import audiocore
import audiobusio

"""
  ----------- User configurable items -----------

  This is where you can customize your display and hardware setup.
  Select either the Pre-Beskar (1.14" LCD) or Beskar (1.3" LCD) display.
  Some lines are commented out which means the line is not active.
  A commented line starts with a #

"""
# Set the display type
DISPLAY        = "Pre-Beskar"           # Adafruit 1.14" LCD display  https://www.adafruit.com/product/4383
#DISPLAY       = "Beskar"               # Adafruit 1.3" LCD display   https://www.adafruit.com/product/4313
DISP_BRIGHT    = 80                     # How bright to make the display - 0% to 100%

# Board being used
BOARD_TYPE     = "ESP32-S3"             # ESP32-S3                  https://www.adafruit.com/product/5477
#BOARD_TYPE    = "FeatherM4"            # Feather M4 Express        https://www.adafruit.com/product/3857
#BOARD_TYPE    = "ItsyBitsyM4"          # ItsyBitsy M4 Express      https://www.adafruit.com/product/3800
#BOARD_TYPE    = "ItsyBitsyRP2040"      # ItsyBitsy RP2040          https://www.adafruit.com/product/4888
#BOARD_TYPE    = "PiPicoRP2040"         # Raspberry Pi Pico RP2040  https://www.adafruit.com/product/4864

# Mandalorian charater sequence that is shown on the display
[... 248 lines cut ...]
            batt_percent = GetBattPercent(vbat_voltage)
        if batt_percent < LOW_BATT_LEVEL :
            # Add low battery icon
            if low_batt_icon == 0 :
                stage.append(batt_tile)
                low_batt_icon = 1
        else :
            if low_batt_icon > 0 :
                # Remove low battery icon
                stage.pop()
                low_batt_icon = 0

    #import supervisor

    #display.brightness = 0
    #splash = display.root_group
    #supervisor.reset_terminal(display.width, display.height//2)
    #splash.y=display.height//2
    #time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 20)
    #alarm.exit_and_deep_sleep_until_alarms(time_alarm, preserve_dios=[digitalio.DigitalInOut(board.D10)])

    #pin_alarm = alarm.pin.PinAlarm(pin=board.D11, value=False, pull=True)
    # Exit the program, and then deep sleep until the alarm wakes us.
    #alarm.exit_and_deep_sleep_until_alarms(pin_alarm)
```

FoamyGuy, 2023-01-21:
@darianbjohnson you can disable things by setting `0` to some variables inside of `mpconfigboard.mk`. 

You can see an example that disables some things here: https://github.com/adafruit/circuitpython/blob/9c066825a7c2127f69da90e80de5eb94e2d92a00/ports/atmel-samd/boards/matrixportal_m4/mpconfigboard.mk#L15-L19

FoamyGuy, 2023-01-21:
@Breazile that may be a different issue I think. Your device is `ESP32S3` and seems to only use 1 display. But this issue is for an S2 device and using two displays. 

It may be best to create a seperate issue for yours instead of posting here.

FoamyGuy, 2023-04-05:
I have replicated this issue on a feather S2 TFT.  I believe the same issue actually effects more ports as well, perhaps all of them. I'm thinking it may trace back to the displayio api change, but haven't done before/after testing yet to narrow it down.

I think @Neradoc saw this occur on Monster M4sk as well, and I've seen the same on that device. 

It seems to boil down to right now the initial setup of 2 displays works fine, but then whenever the device resets it will hard fault, as described in the initial issue post.

I have collected this decoded backtrace from Feather S2 TFT:

```
❯ python tools/decode_backtrace.py adafruit_feather_esp32s2_tft
adafruit_feather_esp32s2_tft
? 0x400A16A1:0x3FFE59A0 0x400A23F2:0x3FFE59C0 0x400A1541:0x3FFE59E0 0x4008307D:0x3FFE5A00 0x4008E2E7:0x3FFE5A20 0x4008E418:0x3FFE5A40 0x400C7CF9:0x3FFE5A60 0x400C7FDF:0x3FFE5A80 0x400CAC5D:0x3FFE5AA0 0x400A0CF7:0x3FFE5AC0 0x400A10AD:0x3FFE5B10 0x400A14C5:0x3FFE5B70 0x400A18B3:0x3FFE5BA0 0x40176990:0x3FFE5BC0
0x400a16a1: reset_cpu at /home/timc/repos/circuitpython/circuitpython/ports/espressif/supervisor/port.c:425
0x400a23f2: reset_into_safe_mode at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../supervisor/shared/safe_mode.c:139
0x400a1541: nlr_jump_fail at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../main.c:1153
0x4008307d: nlr_jump at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../py/nlrsetjmp.c:35
0x4008e2e7: mp_raise_msg at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../py/runtime.c:1626
0x4008e418: mp_raise_ValueError at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../py/runtime.c:1674
0x400c7cf9: common_hal_displayio_display_set_root_group at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../shared-module/displayio/Display.c:410
0x400c7fdf: reset_display at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../shared-module/displayio/Display.c:440
0x400cac5d: reset_displays at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../shared-module/displayio/__init__.c:279
0x400a0cf7: cleanup_after_vm at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../main.c:323
0x400a10ad: run_code_py at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../main.c:475
0x400a14c5: main at /home/timc/repos/circuitpython/circuitpython/ports/espressif/../../main.c:1087
0x400a18b3: app_main at /home/timc/repos/circuitpython/circuitpython/ports/espressif/supervisor/port.c:544
0x40176990: main_task at /home/timc/repos/circuitpython/circuitpython/ports/espressif/esp-idf/components/freertos/port/port_common.c:141
```
I will try to narrow it further and see if I can figure out a fix.

Neradoc, 2023-04-05:
Yep I get a hard crash when saving on the Monster M4sk, with the displays setup form the monster_mask library (which occasionally causes the file I'm editing to be corrupted).
```
You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
NLR jump failed. Likely memory corruption.
Please file an issue with your program at https://github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

FoamyGuy, 2023-04-05:
I followed the backtrace a bit, and I think this actually is the same kind of issue as #7829 is addressing. The circuitpython_spash was being added after already being added to the first I think.

I added a commit to that PR branch that addresses the issue. It's adding the same if statement previously added into the constructor into reset_display() function also.

Tested successfully with the Feather S2 TFT. Will try Monster M4sk next.

Worth noting something may still be wonky though because now I no longer hard crash but I do seem to see duplicated terminal outputs on the screen. First ctrl-C from running results in 3 duplicates stacked one on top of the other. The next ctrl-C to go to reply splits those 3 rows into 2 columns each to make 6 total copies visible.

FoamyGuy, 2023-04-05:
I tried the latest version from that PR on a Monster M4sk and I do still get a hard fault. Slightly different wording I don't have "NLR jump failed. Likely memory corruption."but rather "fault detected by hardware" 

I did ctrl-C instead of saving a file though, that could factor in. 

I'm not sure if it's possible or how to get hard crash traces out of the samd51 / Monster M4sk but if we could it might help work toward the solution. 

Later on the week I'll try to probe a bit further on the feather S2 TFT setup to see if I can figure out the duplicated terminal outputs, and hopefully manage to find whatever hard fault the Monster M4sk is still seeing to get it resolved too.

FoamyGuy, 2023-04-08:
@darianbjohnson if you have a moment please try the latest build from the S3 link for your project that this issue was originally made for. 

I think that the specific hard crash you observed should be resolved after #7829, but I'm also still trying to work through further potential issues. It'd be helpful to know how your setup behaves on the current version.

darianbjohnson, 2023-04-08:
@FoamyGuy I’ll test it out… though I might need a few weeks to retest (heads down on another project that I need to resolve first)

furbrain, 2023-05-15:
@FoamyGuy  I wonder if issue #2204 (fixed by #7983) may be responsible for the hard faults here?

dhalbert, 2023-05-16:
Probably fixed by #7983. Re-test.