adafruit/circuitpython #7475: busio.I2Cslave(responder) seems to be missing

Olzeke51, 2023-01-22:
I'm having issues with my Cytron Maker PI RP2040 timing issues [https://github.com/CytronTechnologies/MAKER-PI-RP2040/issues/5](url)
SO I wanted to use a Adafruit Feather Express M0 as an I2C slave for my HC-SR04 (sonar) unit [robot project]
^
I found this PR [https://github.com/adafruit/circuitpython/pull/1064](url) but that was for version 4 - afaict.
I don't see anything in the 'docs' weblink - just the standard busio.I2C api.
^
Bitbangio was mentioned as a possible alternative - which might help
But was hoping for an already created 'I2Cresponder' api.
^
Any ideas or suggestions will be appreciated.
Gary Olzeke51

Neradoc, 2023-01-22:
You are looking for the `i2ctarget` module (or `i2cperipheral` in CP7).
https://docs.circuitpython.org/en/latest/shared-bindings/i2ctarget/index.html

It is however not available on any M0 build currently, you would have to build it manually by adding `CIRCUITPY_I2CTARGET = 1` in `mpconfigboard.mk` but it's not gonna fit on the trinket unless you disable other modules. It should fit in an express board though.

However, your original issue seems to be a timing problem on RP2040, have you tried 8.0 beta 6 ? Or the latest builds. There have been time issues fixed.

[I fixed the links in your post]

RetiredWizard, 2023-01-22:
I tried your original test program on both a Cyrton Maker PI RP2040 and a Raspberry Pi Pico with version 7.3.3 of Circuitpython and didn't see your timing issue. What version of Circuit Python are you running?

```
Adafruit CircuitPython 7.3.3 on 2022-08-29; Cytron Maker Pi RP2040 with rp2040
>>> import time
>>> for x in range(6):
...     print(x)
...     timestamp2 = time.monotonic_ns()
...     timestamp = time.monotonic()
...     time.sleep(.25)
...     timestamp1 = time.monotonic()
...     print(timestamp1 - timestamp)
...     timestamp1 = timestamp = 0
...     timestamp = time.monotonic_ns()
...     time.sleep(.25)
...     timestamp1 = time.monotonic_ns()
...     print(timestamp1 - timestamp)
...     timestamp1 = timestamp = 0
...
0
0.25
250000000
1
0.25
250000000
2
0.25
250000000
3
0.25
250000000
4
0.25
250000000
5
0.250992
250000000
>>>

```

Olzeke51, 2023-01-22:
Thank you both:
@RetiredWizard 
on the Cytron it was 7.3.3 
BUT
I ran it from code.py not the REPL window - with the Serial monitor on

@Neradoc 
thanks for the link - will follow up on that

RetiredWizard, 2023-01-23:
Strange, I still get good time output:

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.3 on 2022-08-29; Cytron Maker Pi RP2040 with rp2040
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disab                                                                             le.
code.py output:
0
0.25
250000000
1
0.25
250000000
2
0.25
250000000
3
0.250977
250000000
4
0.250977
250000000
5
0.25
250000000

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.3.3 on 2022-08-29; Cytron Maker Pi RP2040 with rp2040
>>>

```

Olzeke51, 2023-01-23:
Thank you - I had considered Mu / Win10 an issue , but the Feather is good !!
I have the I2Cresponder/target information , 
**will consider this closed** - the Cytron issue already has a 'Issue' filed

closed 2023-01-23 by Olzeke51