adafruit/circuitpython #7476: bitstring library needs to be pinned to 3.1.9 for some executables in ports/espressif/esp-idf/esptool_py to function

evildave666, 2023-01-23:
### CircuitPython version

```python
NA
```

### Code/REPL

```python
$ ports/espressif/esp-idf/components/esptool_py/esptool/espefuse.py
```

### Behavior

imports fail due to referencing a class alias that was removed in bitstring library 4.0. The version needs to be forced to 3.1.9 in the build environment.

evildave666, 2023-01-23:
Should probably be addressed upstream but the building document should mention it at least in the meantime.

tannewt, 2023-01-23:
I wonder if we should do one more IDF bump before 8.0 RC. This fix is in the upstream 4.4 branch but ours is too old: https://github.com/adafruit/esp-idf/commits/release/v4.4-circuitpython

dhalbert, 2023-01-23:
> I wonder if we should do one more IDF bump before 8.0 RC. This fix is in the upstream 4.4 branch but ours is too old: https://github.com/adafruit/esp-idf/commits/release/v4.4-circuitpython

I was just going to try merging v4.4 from upstream experimentally to see if it fixed any network or I2C problems. There are a number of interesting commits past where we last merged from.

evildave666, 2023-01-26:
Fixed by https://github.com/adafruit/circuitpython/pull/7486

closed 2023-01-26 by evildave666