adafruit/circuitpython #7481: Can't print single bytes data

musimasami, 2023-01-24:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.0 on 2022-08-18; PCA10059 nRF52840 Dongle with nRF52840
```

### Code/REPL

```python
>>> a = b'abc'
>>> bytes(a)
b'abc'
>>> bytes(a[1:])
b'bc'
>>> bytes(a[1])
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> print(bytes(a[1]))
b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>>
```

### Behavior

Can not handle single bytes data.

elpekenin, 2023-01-24:
When `bytes` receives a number as argument, it will create an empty (all 0s) `bytes` object, with length equal to ve argument received.
Perhaps you wanted to `print(bytes(a)[1])`, ie:
1. Convert the hole thing to a `bytes` object
1. Index the element `1`

What is happening on your code is that `a[1]` is `"b"` which is encoded as `98` (you can check this by doing `ord("b")`) Thus, you are creating a `bytes` instance with that length

PS: Tested this on Windows' Python, but I don't think CircuitPy wants/does a different implementation of the standard functions/classes

dhalbert, 2023-01-24:
@elpekenin is correct, this is how Python works. The elements of a `bytes` objects are integers, as opposed to single-element bytes. This is different than how strings work.
```py
>>> a = b'abc'   # bytes
>>> a[1]    # get a single element, which is an integer
98
>>> a[1:2]  # get a sub-bytestring
b'b'

>>> s = 'abc'    # string
>>> s[1]  # get a single element, which is another string
'b'
```

closed 2023-01-24 by dhalbert

musimasami, 2023-01-25:
I wanted to extract one character as bytes. Thank's
>>> a = b'abc'
>>> b = a[:1]   # extract one character as bytes
>>> c = a + b
>>> c
b'abca'
>>> b
b'a'
>>> a
b'abc'
>>>