adafruit/circuitpython #7482: Documentation for traceback.format_exception does not match implementation

dannystaple, 2023-01-24:
### CircuitPython version

```python
Adafruit CircuitPython 7.2.0 on 2022-02-24; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico
```

### Code/REPL

```python
import traceback

try:
    raise RuntimeError("Hello")
except Exception as e:
    print("Formatted was")
    print(traceback.format_exception(e)
    raise
```

### Behavior

```python
>>> %Run -c $EDITOR_CONTENT
Formatted was
Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
TypeError: 'value' argument required
```

### Description

- Documentation at https://docs.circuitpython.org/en/latest/shared-bindings/traceback/index.html#traceback.format_exception suggests that the parameters `value` and `tb` are `value: Optional[[BaseException](https://docs.python.org/3/library/exceptions.html#BaseException)] = None, tb: Optional[[types.TracebackType](https://docs.python.org/3/library/types.html#types.TracebackType)] = None`.
- However, the tests explicitly provide these, and the above errors (and a similar one for TB if value is provided) always shows.

Neradoc, 2023-01-24:
Hi, the syntax with a single argument (brrowed from Python 3.10) was added in Circuitpython 8.0.0.
There's a menu at the bottom left of the documentation where you can select the version or Circuitpython that you can the docs for:
https://docs.circuitpython.org/en/7.2.x/shared-bindings/traceback/index.html

jepler, 2023-01-25:
Thanks for clarifying that @Neradoc !

closed 2023-01-25 by jepler