adafruit/circuitpython #7487: Firmware.bin and CircuitPython.bin are different

BrainBoardz, 2023-01-26:
Hi There: 

I've developed an ESP32-S3 board and have been compiling circuit python (7.3.3 etc.) for it as per the instructions. When I  do a compile I get a firmware.bin, circuitpython-firmware.bin and firmware.uf2 files. I've used the ESP Web Serial tool to upload my firmware. For some reason firmware.bin works perfectly (in Mu CP code runs perfectly). However, if I use the same ESP Web Serial tool and upload circuitpython-firmware.bin to my board it doe not work and the board goes into a endless loop and the CIRCUITPYTHON drive never mounts. I've also noticed the size of the two firmware files are slightly different.  The CB bin is a bit smaller (1,383KB vs 1,447KB).  I can't think of any reason for this.  Does anyone have any suggests. Is this normal? Should I just be using firmware.bin instead? Any help would be appreciated.

 I have also be unable to put the UF2 on my board (it just gets put on the "drive" if I drag the UF2 to the board), but I think this may be because I do not have a UART chip on my board or that I have to use ESPTOOL for that process.

dhalbert, 2023-01-26:
Use `firmware.bin`. `circuitpython-firmware.bin` is not combined with the proper second-stage bootloader.

To install the UF2 bootloader, you need a build of https://github.com/adafruit/tinyuf2 for your board. If your board is straightforward, you may just be able to use an existing build for, say, one of the Espressif ESP32-S3 dev boards. To install TinyUF2, follow this procedure, more or less, but substitute the TinyUF2 build you have: https://learn.adafruit.com/adafruit-metro-esp32-s2/install-uf2-bootloader (note these instructions are for an S2 board).

This support question is not really a GitHub issue: a better place is our discord, https://adafru.it/discord, in #circuitpython-dev or #help-with-circuitpython, or in the forums: https://forums.adafruit.com/viewforum.php?f=60.

BrainBoardz, 2023-01-26:
Thanks for the fast feedback. Good to know that firmware.bin is the correct one! I'll give the UF2 a go and I will post on discord or in the forums from here on.

closed 2023-01-26 by dhalbert

jepler, 2023-01-27:
Additionally, you can see the commandline that circuitpython thinks is right for flashing each file:
flashing `firmware.bin` at 0:
```
$ make BOARD=adafruit_feather_esp32s3_4mbflash_2mbpsram -n flash
esptool.py --chip esp32s3 -p /dev/serial/by-id/usb-Espressif_ESP32-S2_0-if00 --before=default_reset --after=no_reset --baud 921600 write_flash --flash_mode qio --flash_freq 40m --flash_size 4MB 0x0000 build-adafruit_feather_esp32s3_4mbflash_2mbpsram/firmware.bin
```

flashing circuitpython-firmware.bin at 0x10000:
```
esptool.py --chip esp32s3 -p /dev/serial/by-id/usb-Espressif_ESP32-S2_0-if00 --before=default_reset --after=no_reset --baud 921600 write_flash --flash_mode qio --flash_freq 40m --flash_size 4MB 0x10000 build-adafruit_feather_esp32s3_4mbflash_2mbpsram/circuitpython-firmware.bin
```

BrainBoardz, 2023-01-30:
Thanks for all the help with this. I haven't tried flashing a build directly before, so I will give that a try!