adafruit/circuitpython #7488: [Classroom Development] Displayio locks SPI bus

demc, 2023-01-26:
### CircuitPython version

```python
7.3.3
```

### Code/REPL

```python
import board
import time
import terminalio
import displayio
import busio
from adafruit_display_text import label
from adafruit_st7789 import ST7789
from xpt2046 import Touch
from digitalio import DigitalInOut, Direction

# Release any resources currently in use for the displays
displayio.release_displays()

TFT_WIDTH = 320
TFT_HEIGHT = 240

tft_dc = board.GP8
tft_cs = board.GP9
spi_clk = board.GP10
spi_mosi = board.GP11
spi_miso = board.GP12 # added this
tft_rst = board.GP15 # changed this
backlight = board.GP13
spi = busio.SPI(spi_clk, spi_mosi, spi_miso)

display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=tft_rst)

display = ST7789(
    display_bus,
    width=320,
    height=240,
    rotation=90,
    backlight_pin=backlight
)

touch_cs = board.GP16

touch_x_min = 164
touch_x_max = 1826
touch_y_min = 164
touch_y_max = 1928

touch = Touch(spi, cs=touch_cs,
              width=320, height=240,
              x_min=touch_x_min, x_max=touch_x_max,
              y_min=touch_y_min, y_max=touch_y_max)

group = displayio.Group()
display.show(group)
```

### Behavior

`displayio.FourWire` locks the SPI bus. The `xpt2046` driver can not talk to the board because they share the same SPI bus.

### Description

I am educator working on curricula using Pico + CircuitPython. 

I've purchased a class set of waveshare lcd screens: https://www.waveshare.com/wiki/Pico-ResTouch-LCD-2.8#Arduino

I am by no means an expert but after reading #1760 it appears that there are limitations with the existing implementing of `FourWire` that effectively cripples this hat. 

Here's what I've observed:

1. The ST7789 is able to work fine in isolation. I can render things on the screen.
2. The xpt2046 is able to work fine in isolation. I can read touch events reliably from the screen.
3. Combining `displayio` with xpt2046 doesn't work because I can not use the LCD and Touch chip select pins as laid out in the wiring of the waveshare device. 

Looking closely at the [waveshare docs](https://www.waveshare.com/wiki/Pico-ResTouch-LCD-2.8#Arduino), I've gathered that both the  ST7789 and xpt2046 are sharing the same SPI bus. I can toggle either device by setting `GP9` (LCD_CS) or `GP16` (touch CS). As best as I can tell, toggling those pins configures which device is transmitting information on the SPI bus (correct me if I'm wrong or naive).

Trying to manually toggle `GP9` to momentarily disable the LCD screen so I can read from the touch screen does not appear to work because I get an error about the pin being in use (I get why this restriction exists, but it's really weird). 

I am open to any suggestions! If I can't figure out how to make this work, I will have to rewrite my curricula for MicroPython because of the restrictions imposed by CircuitPython.

As an aside:

It was mentioned in #1760 that I could probably use the RGB display driver instead of displayio. I'm not sure if that is suitable because I need to be able to render text, which, as far as I can tell, is only possible using displayio. 

Thank you!

Neradoc, 2023-01-26:
You don't need to toggle the pins manually, the drivers do that on their own. What is that `xpt2046` module your are importing ?
Note: This looks like a support issue, they are better handled on the forums or the Adafruit discord.

deshipu, 2023-01-26:
The drivers should handle the CS pins internally, you don't need to toggle them yourself. It should just work. The #1760 is really about sharing the D/C and reset pins between two or more displays.

demc, 2023-01-26:
Thanks for the quick reply!

I am using an adaptation of [this](https://github.com/rdagger/micropython-ili9341/blob/master/xpt2046.py)  MicroPython driver. Waveshare did not publish their own native driver. 

Thank you for pointing me to Discord. I will also try asking this there. 

Pertaining to hats that use a single SPI bus for both the display and other peripherals, is that actually supported by displayio?

Are there any additional resources I can read about this?

demc, 2023-01-26:
@Neradoc very kindly helped me figure this out on Discord. Closing.

I will publish the combined driver when I get all the peripherals working perfectly.

closed 2023-01-26 by demc