adafruit/circuitpython #7490: More debug info (maybe core dump) on panic

tannewt, 2023-01-26:
We see hard faults after a while on ESP and they are difficult to debug. The IDF can [produce core dumps](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/core_dump.html). Perhaps we should set those up so we have more debug info after the fact.

It may require changing the partition table. We can do this along with IDF 5.0 because it may require partition changes too. We may be able to shrink the TinyUF2 partition too.

bill88t, 2023-01-27:
I'd like to argue openocd & gdb with breakpoints & line by line execution cover most use cases of coredumps.

dhalbert, 2023-01-27:
> I'd like to argue openocd & gdb with breakpoints & line by line execution cover most use cases of coredumps.

Our aim here is to allow any user to send us a core dump after a hard crash, without having to do any set up. For our own purposes, yes, interactive debugging is great.

d-c-d, 2023-01-27:
Are there any ideas that can be gleaned from [Memfault's collection and debug tools for core dumps](https://docs.memfault.com/docs/mcu/coredumps/)?

dhalbert, 2023-01-27:
This is a specific [feature](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/core_dump.html) of ESP-IDF. The "core dump" is not a full RAM dump, but an assortment of stuff that includes stack backtraces, registers, task info, etc. It's really more of a "state" dump.

d-c-d, 2023-01-28:
Thanks - there are a lot of other ideas that [Memfault](https://docs.memfault.com/docs/mcu/introduction) uses  to assist developers in collecting data besides full device coredumps; such as backtraces, debug registers, logs, and custom metrics. ( I have nothing to gain, but have learned from their [interrupt blog](https://interrupt.memfault.com/blog/).

tannewt, 2023-04-14:
Here is an issue this would be nice for: https://github.com/adafruit/circuitpython/issues/7405