adafruit/circuitpython #7498: PewpewM4 Serial Connection Broke Somewhere between 7.0.0 and 7.1.0

FoamyGuy, 2023-01-28:
### CircuitPython version

```python
Adafruit CircuitPython 7.1.0-beta.0 on 2021-11-12; PewPew M4 with samd51g19
Board ID:pewpew_m4
```

### Code/REPL

```python
N/A. code.py doesn't seem to make any difference.
```

### Behavior

When connected via USB there is no serial connection that becomes available. i.e. `/dev/ttyACM0` never appears. 

### Additional information

On version `7.0.0` the serial connection is made and `/dev/ttyACM0` does appear.

On versions `7.1.0` and newer the serial connection is never made. `/dev/ttyACM0` never appears.

I started trying to bisect the few beta / rc releases between those versions to nail it down to a specific release, but had trouble with my computer during the attempts. I will try to finish that up in the next day or two and post back to here with the specific version where this breaks.

FoamyGuy, 2023-01-28:
I've narrowed it further to between these two versions:

Up to this version the serial connection behaves normally:
```
Adafruit CircuitPython 7.1.0-beta.1 on 2021-11-30; PewPew M4 with samd51g19
Board ID:pewpew_m4
```

Starting with this version the serial connection no longer appears when the device turns on. 
```
Adafruit CircuitPython 7.1.0-beta.2 on 2021-12-10; PewPew M4 with samd51g19
Board ID:pewpew_m4
```

RetiredWizard, 2023-01-28:
The only SAMD board I have is the Sparkfun SAMD51 Thing Plus which boots 8.0.0 so I'm wondering if the issue is caused by one of the PewPewM4 changes. Have you tried backing out these two changes?

[image]

RetiredWizard, 2023-01-28:
After looking at my post, the board_deinit was added to most if not all the SAMD boards and the second change only disables the ALARM so probably not related....

FoamyGuy, 2023-01-28:
Ah, I think it would be the `CIRCUITPY_USB_CDC = 0` that would do it. 

I made a build with that line removed and I do have serial connection again using that build. Perhaps this was an intentional change.

FoamyGuy, 2023-01-28:
If I am understanding the history correctly I think this commit is where it was first added: https://github.com/adafruit/circuitpython/commit/4cf6579e2201e62988fbb6cf204037ce0d80e3f1

FoamyGuy, 2023-01-28:
That commit leads back to here I think: https://github.com/adafruit/circuitpython/pull/4333

Which does look like an intentional change to save space. 

Thank you @RetiredWizard! the diffs you posted got me pointed in the right direction to understand this. 

I can make my own build with serial enabled for my purposes.

closed 2023-01-28 by FoamyGuy

dhalbert, 2023-01-28:
@deshipu Did you want to turn off CDC?

deshipu, 2023-01-28:
@dhalbert no, I don't think I did, we might be able to switch off something else if there is no room