adafruit/circuitpython #7500: USB_CDC Serial console doesn't work stable , loop back test failed. (Both ESP32-S3 and Raspberry pi Pico)

zawakenjp, 2023-01-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Raspberry Pi Pico with rp2040
Board ID:raspberry_pi_pico

Adafruit CircuitPython 7.3.3 on 2022-08-29; ESP32-S3-DevKitC-1-N8R2 with ESP32S3
Board ID:espressif_esp32s3_devkitc_1_n8r2
```

### Code/REPL

```python
while 1:
    data=input()
    print(data)
```

### Behavior

I faced unstable USB_CDC console communication.
Then I performed simple loopback test triggered from raspberry pi 3b+ to both ESP32-S3 and Raspberry pi Pico.
And I found both of them failed loopback test. I suspect it would be a bug.
Please check whether it is bug or not.

ESP32-S3 : freeze at  644 cycle
643 abc12345
644 abc12345

Raspberry pi Pico: freeze at 96 cycle
95 0;🐍code.py | 8.0.0-beta.6abc12345
96 0;🐍code.py | 8.0.0-beta.6abc12345

### Description

This loopback test is very simple.
Triggered from Raspberry pi 3b side.
1) Raspberry pi 3b sends just 1 line.  
2) Circuitpython side(EPS32-S3 or Pico) will receive and return same value.
3) Raspberry pi 3b will receive 1 line and resend same value.
4) loop 2)-4) 

Raspberry pi 3b+ side program:

import serial
_serial = serial.Serial('/dev/ttyACM0')
data = b'abc12345\r\n'
print("tx: ", data)
_serial.write(data)
i=0
while 1:
    data = _serial.readline()
    print (i,data.decode('utf-8'))
    #print (i,data)
    _serial.write(data)
    i=i+1

Neradoc, 2023-01-28:
Can you elaborate on the "instability" you encountered in your real use case ?

Your test here has a problem, since `input()` is for interactive use it echoes the characters it receives, so the board sends the line twice. And your test code will be reading only one line before sending another. You can show it by sending `i` and seeing what you read as reply. You'll receive each line in double, with the number being half of the value of i.

```py
data_out = b'LINE\r\n'
i = 0
while True:
	_serial.write(f"{i} ".encode() + data_out)
	data_in = _serial.readline()
	print(f"IN : {i:4d} = {data_in}")
	i = i + 1
```

Testing your code on my mac doesn't freeze, but on a Pi it does stop after a few hundred loops. Since it only reads half of what the board sends, it might be filling up some buffer.

You can use `usb_cdc.console` instead, or enable the `usb_cdc.data` port (on the pico at least, the ESP has less endpoints available, so you would have to disable something). This returns bytes, not a string, with the line endings in it.
```py
"""code.py"""
import usb_cdc
while True:
    data = usb_cdc.console.readline()
    usb_cdc.console.write(data)
```

On a board without native USB, you can use stdin. However, maybe because it's a text stream, it seems that stdin converts \r into \n, so we have the same problem of the board sending 2 lines (one empty) per "line" received. So here I make it ignore the empty line.
```py
"""code.py"""
import sys
while True:
    data = sys.stdin.readline()
    if data.strip():
        print(data.strip()) # or sys.stdout.write(data)
```

dhalbert, 2023-01-28:
> `95 0;code.py | 8.0.0-beta.6`

This extra stuff is from the status bar, which you can disable. Is that something that's interfering?

zawakenjp, 2023-01-28:
Neradoc, thank you for your support.
> Can you elaborate on the "instability" you encountered in your real use case ?

I use Raspberry pi PICO for controlling Neopixel (900pcs LED, 1 channel) via USB_CDC console from Raspberry pi 3B+(or 4B) serial function node on node-red 3.0.2.
I use Neopixel as a display, therefore long time stable work needed at least 12 hours.
I face 1 or 2 times freeze in one day.
(Fortunately, Neopixel update cycle is not so frequent. About 10 times / 3 seconds. )    

I changed PICO into ESP32-S3 and it works more stably, but ESP32-S3 also failed loopback test. 

As you mentioned, I tried USB_CDC.data , it works well on more than 7 hours loopback test.
I'll use USB_CDC.data insted of input()/print() temporally.

zawakenjp, 2023-01-29:
dhalbert, I disabled status bar and tried loopback test.
I found extra stuffs were gone, but same output as ESP32-S3. 
I think status bar does not cause instability.

PICO : freeze at xxx cycle
xxx abc12345
xxx abc12345

zawakenjp, 2023-01-30:
I tried USB_CDC.data and USB__CDC.console both.

```
serial = usb_cdc.data
while 1:
    data=serial.readline()
    serial.write(data)
```
```
serial = usb_cdc.console
while 1:
    data=serial.readline()
    serial.write(data)

```

PICO:
  USB_CDC.data -> works very stable more than 24hours 
  USB_CDC.console ->  unstable,  I got a message of Reloading.
  
  I can use USB_CDC.data for my purpose but I cannot use USB_CDC.console  insted of input()/print().

> 66731 abc12345
> 66732 abc12345
> 66733 abc1234abc12345
> 66734 bc12345
> 66735 abc12345
> 66736 abc12345
> 66737 
> 66738 Code stopped by auto-reload. Reloading soon.
> 66739 
> 66740 オートリロードがオンです。ファイルをUSB経由で保存するだけで実行できます。REPLに入ると無効化します。
> 66741 code.py 出力:

dhalbert, 2023-03-21:
@rena2019 The Espressif chips don’t have enough USB endpoints to have both CDC console and data, unless you turn everything else off. See https://learn.adafruit.com/customizing-usb-devices-in-circuitpython?view=all#hardware-limitations-3096627, especially toward the end.

dhalbert, 2023-10-25:
Closing due to inactivity. Please reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert