adafruit/circuitpython #7507: readthedocs builds are failing for latest 8.0.0 builds

dhalbert, 2023-01-29:
From readthedocs logs:
```
git clone --no-single-branch --depth 50 https://github.com/adafruit/circuitpython .
git checkout --force origin/main
git clean -d -f -f
asdf global python 3.11.0
python -mvirtualenv 
python -m pip install --upgrade --no-cache-dir pip setuptools<58.3.0
python -m pip install --upgrade --no-cache-dir pillow mock==1.0.1 alabaster>=0.7,<0.8,!=0.7.5 commonmark==0.9.1 recommonmark==0.5.0 sphinx sphinx-rtd-theme readthedocs-sphinx-ext<2.3
python -m pip install --exists-action=w --no-cache-dir -r requirements-doc.txt

python tools/ci_fetch_deps.py docs HEAD
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/circuitpython/checkouts/latest/tools/ci_fetch_deps.py", line 138, in <module>
Target: docs
    main()
  File "/home/docs/checkouts/readthedocs.org/user_builds/circuitpython/checkouts/latest/tools/ci_fetch_deps.py", line 93, in main
    raise RuntimeError(f"Unsupported target: {TARGET}")
RuntimeError: Unsupported target: docs
```
I think this may be due to the latest `ci_fetch_deps.py` changes.

Tagging @MicroDev1

dhalbert, 2023-01-29:
[image]

MicroDev1, 2023-01-30:
The following changes are needed:
```patch
- python tools/ci_fetch_deps.py docs HEAD
+ python tools/ci_fetch_deps.py build-doc
```
Add the following if tags are needed.
```bash
git fetch --no-recurse-submodules --shallow-since="2021-07-01" --tags https://github.com/adafruit/circuitpython HEAD
git fetch --no-recurse-submodules --shallow-since="2021-07-01" origin HEAD
git repack -d
```

jepler, 2023-01-30:
> git repack -d

curious why it's worth spending time repacking when not many git operations will be done subsequent to this.

MicroDev1, 2023-01-30:
> curious why it's worth spending time repacking when not many git operations will be done subsequent to this.

It exits immediately with the message `Nothing new to repack`.
See: https://stackoverflow.com/questions/63878612/git-fatal-error-in-object-unshallow-sha-1#comment118418373_63879454

dhalbert, 2023-01-30:
I am not sure we need tags for the RTD build. I don't know how to check that without proposing just the first change and then merging it.

closed 2023-01-30 by MicroDev1

dhalbert, 2023-01-30:
@MicroDev1 Thank you -- that one-line fix worked:

[image]

PR #7508 (2023-01-30): Fix broken RTD builds
Fixes #7507, based on @MicroDev1 's suggestions in that issue.