adafruit/circuitpython #7515: scorpio: spurious USB resets when running demo code

jepler, 2023-01-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit Feather RP2040 Scorpio with rp2040
```

### Code/REPL

```python
# grab the code sample from https://learn.adafruit.com/introducing-feather-rp2040-scorpio/using-adafruit_neopxl8
```

### Behavior

Sometimes the USB device will disconnect and reconnect.

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
157.9fps
154.2fps
159.0fps
156.6fps

[tio 16:28:51] Disconnected
[tio 16:28:52] Connected
[tio 16:28:53] Disconnected
[tio 16:28:54] Connected
156.7fps
156.4fps
158.7fps
154.6fps
156.3fps
156.6fps
```

### Description

Compared to most boards / exampes, this is using two less used features: `StateMachine.background_write` and `bitops.bit_transpose`.

### Additional information

This or something like it has been seen by me, @jedgarpark and @djecken as discussed on slack. I'm on a prerelease board while they have store revisions.

dhalbert, 2023-02-03:
> Compared to most boards / examples, this is using two less used features: `StateMachine.background_write` and `bitops.bit_transpose`.

Would either of these spend time in a tight loop and freeze out USB task handling?

Do you feel this is a showstopper for 8.0.0?

jepler, 2023-02-03:
> Would either of these spend time in a tight loop and freeze out USB task handling?

I looked for that, as it's the first explanation that occurred to me too.

`background_write` has one place where it loops (based on the status of an ongoing DMA transaction), but it calls RUN_BACKGROUND_TASKS every loop.

`bit_transpose` is a tight loop, but it runs with interrupts enabled:
```
static void bit_transpose_8(uint32_t *result, const uint8_t *src, size_t src_stride, size_t n) {
    for (size_t i = 0; i < n; i++) {
        transpose_8(result, src, src_stride);
        result += 2;
        src += 1;
    }
}
```
but for the amount of data being processed (8 strands of 30 LEDs at 24bpp), the time is only ~100us.

Another possible explanation would be electrical, though I don't like it: the PIO program, when running, causes 8 outputs to switch at the same time, and with wiring problems that can amount to ~400mA being switched on and off. something ground bounce something words an EE would say?

Just running the `bitops.bit_transpose` in a loop did not reproduce the problem over the course of 5 minutes.

jepler, 2023-02-08:
Based on internal discussions, this is also affecting Arduino. Internal testing indicates that reducing the drive strength on the #16-23 pins to 2mA seems to fix it.

PR #7558 (2023-02-09): Use lowest drive level for PIO
This appears to fix the scorpio problem (closes #7515) and only mildly affected a direct-drive LED example.
closed 2023-02-09 by jepler