adafruit/circuitpython #7516: IS31FL3741 Native Driver Incorrect Display Output on 8.x

FoamyGuy, 2023-01-30:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-rc.1 on 2023-01-30; Adafruit LED Glasses Driver nRF52840 with nRF52840

and

Adafruit CircuitPython 7.3.3 on 2022-08-29; Adafruit LED Glasses Driver nRF52840 with nRF52840
```

### Code/REPL

```python
Example code from here:
https://github.com/adafruit/Adafruit_CircuitPython_IS31FL3741/blob/main/examples/is31fl3741_native.py
```

### Behavior

On CircuitPython 7.3.3 the example program runs as expected producing yellow scrolling text and blue ring animations on the glasses. 

On CircuitPython 8.0.0-rc.1 The same program produces different results. The blue ring animations seem to work mostly as expected, but the text is not rendering correct. It's getting all jumbled up and different colors. You can tell it's drawing "something" and moving it by changing it's x position, but the thing it draws is not recognizeable to me. It looks lik e the pixels got jumbled up or corrupted during communication to the driver chip possibly.

### Additional information

I will work on bisecting versions to try to figure out more specifically when this changed between 7 and 8. Right now I've only tried 7.3.3 and 8.0.0-rc.1

FoamyGuy, 2023-02-02:
I have narrowed the version a bit further. `8.0.0-beta.1.uf2` which is the first build linked in S3 after `7.3.3` exhibits this behavior as well. 

[There are a fairly large amount of commits](https://github.com/adafruit/circuitpython/compare/7.3.3...adafruit:circuitpython:8.0.0-beta.1) between those two. In the coming days I'll work on bisecting between them to try to find the commit where the behavior changes.

gamblor21, 2023-02-02:
I did not have long to look at this but I am wondering if there is something with the memory allocation that changes (in `displaycore.c` I think?). I was(am) a rookie about memory allocation for CircuitPython when I did that and wrote that and mostly copied it from the RGBMatrix code. Just an idea of where you could look first.

PR #7546 (2023-02-04): fix for native is31fl3741
resolves #7516 

Tested successfully with:
```
Adafruit CircuitPython 8.0.0-rc.2-dirty on 2023-02-03; Adafruit LED Glasses Driver nRF52840 with nRF52840
Board ID:adafruit_led_glasses_nrf52840
UID:4D43A3F73BE5AE3A
```
closed 2023-02-04 by gamblor21