adafruit/circuitpython #7523: When connected to a mac, the RPi Pico is stuck in boot loop

abhnvahoo, 2023-01-31:
### CircuitPython version

```python
Adafruit CircuitPython 7.3.0; Raspberry Pi Pico with rp2040
```

### Code/REPL

```python
import board
import digitalio
import time

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

### Behavior

BOOTSEL mode works fine. I am able to flash the UF2 file.

But after reboot, the flash storage is mounted, then unmounted. It goes in a loop for a few times before silently stopping. MacOS throws numerous "This device may not have been ejected properly". I don't see anything in Serial Console or in macOS' kernel logs.

I have tested it on Windows/Linux. It works as intended with the same cable.

### Description

P.S.: I have also tried flashing CircuitPython and adding example code on Windows. It works fine there but on mac, I face the same issue. The LED blinking works but no success on accessing with the system.

anecdata, 2023-01-31:
Does it happen if you flash nuke the board, then re-install UF2? Does it happen in 7.3.3? 8.0.0-rc.1?

abhnvahoo, 2023-01-31:
Yes it is mac-centric. In more than an year of working on Win/Linux with the same kit, I haven't faced such an issue.

I nuked using storage.erase_filesystem(), tried 7.3.3 & 8-rc1. Still facing the same issue. It mounts the storage. I can see my code files but the drive disappears very quickly.

anecdata, 2023-01-31:
It doesn't sound much like a board issue, but there's a more thorough flash erase here: <https://learn.adafruit.com/getting-started-with-raspberry-pi-pico-circuitpython?view=all#circuitpython>. Not too many people on Ventura 13.2 yet, do yo have access to a 13.1 Mac?

abhnvahoo, 2023-01-31:
Tried the nuke.cf2. It didn't help. I don't have Ventura 13.1. I'll try if I can find a friend who still has not updated.
It's just one Pico connected to the mac at any time. I created a fresh new user on my system. So my user profile or configuration is not an issue.

dhalbert, 2023-01-31:
I tested with your test program on a 2017 Intel Macbook, and an M1 Mac Mini, both running Ventura 13.2, and I did not encounter this problem.

Are you using a USB hub, by any chance? I was directly connected to both (with a USB-A to USB-C adapter on the Macbook).

abhnvahoo, 2023-01-31:
> Are you using a USB hub, by any chance? I was directly connected to both (with a USB-A to USB-C adapter on the Macbook).

I am using a USB3 Hub but I have also tested it on PC. Your comment really pins the issue down to either the hub or my machine. I have ordered a USB-C to microUSB. After I get it, I will update in a day.

I had a bit of a win a few minutes ago though. I found that if I connect to the Serial Console in the first 5 seconds, it works beautifully as long as I am connected to the console. Everything from file-system access to auto-reloading is working fine. If I fail to connect to serial console or exit the console, the reported behaviour resumes.

dhalbert, 2023-01-31:
I have an Amazon Basics USB-C-to-USB-A USB3 hub which is is a poor performer.  I was using it on Linux. The aluminum-cased "all-in-one" adapter I have works better.  It has 3 USB-A's, HDMI, Ethernet, USB-C power input, etc.

aedile, 2023-04-04:
FWIW, I'm having a similar problem.  I was on Monterey 12.5.1 and all CircuitPython devices are no longer recognized.  I'm mostly using Waveshare RP2040 LCD boards.  I do not see the same mount/unmount behavior to my knowledge but no CircuitPython devices ever show up.  This after weeks of dev with no issues.  I also have a mac Catalina 10.15.7.  When my boards stopped working there, I moved over to that one.  After less than an hour of working, the same thing happened.  I think it may be something to do with the way I'm doing my workflow of flashing and rebooting the device.  The mac always complains about an improper ejection.  It's not the device itself.  I can plug it into a windows PC and it works fine.  It's not even the mac itself.  I can put the device into boot mode and it's detected.  I can flash it with Micropython uf2 or a custom C boot image I've made/compiled myself.  All of these work.  But once I switch back to the CircuitPython uf2 (any version I can find, BTW), the system refuses to read the board as a drive.  I suspect that the OS is seeing shenanigans with the device behaving strangely (still get those "is this a keyboard popups") and puts it on a naughty list and refuses to read it.  I have yet to find any evidence for this but it fits the facts and I'm searching.  My intent was to submit my own bug but after careful review of existing issues, I think this one might be the same thing.  If you'd like to ask questions about different environments in the Mac ecosystem and/or want to look at logs, please respond here and I'll get you what I can.

aedile, 2023-04-05:
https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#erase-circuitpy-without-access-to-the-repl-3105309-32

I followed the instructions to enter safe mode on the above link and deleted my code.py while in safe mode (drive mounted properly in safe mode).  Once I did that I was able to see the drive when I plugged it back in.  I haven't had time to try a new code.py file yet but will report back when I do.  @abhnvahoo  - have you tried safe mode yet?

jess1447, 2023-04-06:
And with a new code.py, the procedure above seems to have fixed whatever issue I was facing.  Not sure it'll fix yours but I've repeated it across all my boards and it seems to have worked.  No clue why it'd read on a mac for a time or two then fail.  No clue why it seemed to work on its own power or on a windows machine.  But it seems to be working now.

dhalbert, 2023-10-25:
Closing due to inactivity. Please reopen if this is still an issue for you.

closed 2023-10-25 by dhalbert