adafruit/circuitpython #7528: Please support the PicoPi with PS2IO

werpu, 2023-01-31:
Just as the title says.... I have been running into this issue.

werpu, 2023-02-08:
Thanks for categorizing it... And thanks for having it on your todo list now!
I really appreciate it!

jepler, 2023-05-24:
Example of doing this in raw python code (micropython): https://raspico.blogspot.com/2022/05/using-pio-to-interface-ps2-keyboard.html

This could be used as a guide for implementing ps2io in "plain circuitpython" or for creating a C implementation in the core.

There's also an implementation of the related IBM PC keyboard protocol in https://learn.adafruit.com/ibm-pc-keyboard-to-usb-hid-with-circuitpython/overview

tannewt, 2023-05-25:
Here is the SAMD implementation: https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/common-hal/ps2io/Ps2.c