adafruit/circuitpython #7543: Only 3 sockets available on raspberrypi with web workflow disabled

anecdata, 2023-02-03:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-rc.1 on 2023-01-30; Raspberry Pi Pico W with rp2040
(but same all the way back to the first raspberrypi build)
```

### Code/REPL

```python
import time
import wifi
import socketpool
from secrets import secrets

time.sleep(2)  # wait for serial
wifi.radio.connect(secrets['ssid'], secrets['password'])
time.sleep(0.1)  # wait for raspberrypi IPv4 to become valid
print(f"{wifi.radio.ipv4_address}")

pool = socketpool.SocketPool(wifi.radio)

socks = []
while True:
    socks.append(pool.socket(pool.AF_INET, pool.SOCK_DGRAM))
    print(f"{socks}")
```

### Behavior

```
[tio 15:56:54] Connected
192.168.6.205
[<Socket>]
[<Socket>, <Socket>]
[<Socket>, <Socket>, <Socket>]
Traceback (most recent call last):
  File "code.py", line 15, in <module>
RuntimeError: Out of sockets
```

### Description

With web workflow disabled, `espressif` correctly shows 8 sockets.
I thought `raspberrypi` had 4 sockets available.

I'm also not sure how web workflow has enough sockets: https://github.com/adafruit/circuitpython/issues/7333#issuecomment-1349491317

dhalbert, 2023-02-03:
This three-socket limit is true only of `SOCK_DGRAM` sockets. If I remove `pool.SOCK_DGRAM`, I can create five sockets (without web workflow).

anecdata, 2023-02-03:
That's curious, I went with `DGRAM` assuming it would have the lowest resource requirements.

dhalbert, 2023-04-14:
@anecdata is this still an issue for you? There was a timer fix that helped this to some extent, we seem to remember.

anecdata, 2023-04-14:
`Adafruit CircuitPython 8.1.0-beta.1-23-g0aea45516 on 2023-04-12; Raspberry Pi Pico W with rp2040`
Reset between testing each type of socket.

Web Workflow not enabled, same as before:
5 `SOCK_STREAM`, then `RuntimeError: Out of sockets`
3 `SOCK_DGRAM`, then `RuntimeError: Out of sockets`
4 `SOCK_RAW`, then `RuntimeError: Out of sockets`

Web Workflow enabled, same as before:
5 stream
2 dgram
4 raw

<details>

<summary>(test code...)</summary>

```py
import wifi
import socketpool
import microcontroller
import traceback
import random
from secrets import secrets

time.sleep(3)  # wait for serial
while not wifi.radio.connected:
    try:
        wifi.radio.connect(secrets['ssid'], secrets['password'])
    except Exception as e:
        traceback.print_exception(e, e, e.__traceback__)
        time.sleep(1)
print(f"{wifi.radio.ipv4_address}")

pool = socketpool.SocketPool(wifi.radio)

socks = []
sock_types = {pool.SOCK_STREAM: "pool.SOCK_STREAM", pool.SOCK_DGRAM: "pool.SOCK_DGRAM", pool.SOCK_RAW: "pool.SOCK_RAW"}
sock_type = random.choice((pool.SOCK_STREAM, pool.SOCK_DGRAM, pool.SOCK_RAW))
while True:
    try:
        socks.append(pool.socket(pool.AF_INET, sock_type))
        print(f" {sock_types[sock_type]} {socks}")
    except Exception as e:
        traceback.print_exception(e, e, e.__traceback__)
        time.sleep(1)
        break
microcontroller.reset()
```

</details>

jepler, 2023-06-21:
There's going to be some resource limit. If this is causing a problem with a specific use-case then it may be worth discussing it further.

closed 2023-06-21 by jepler