adafruit/circuitpython #7557: wifi.radio.connect() with timeout vs non-responding DHCP server

vladak, 2023-02-08:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0 beta 3, QtPy with ESP32-S2
```

### Code/REPL

```python
import wifi

wifi.radio.connect(secrets["ssid"], secrets["password"], timeout=10)
```

### Behavior

In case the DHCP server is not responding to the DHCP requests sent via the `connect()` call, the call does not return after 10 seconds - the timeout seems to be is ignored (assuming the timeout is specified in seconds). 

My experiment lasted for about 15 minutes, perhaps there is some other timeout that may kick in much later on.

### Description

Besides the timeout not having any effect in this case, the documentation on https://docs.circuitpython.org/en/latest/shared-bindings/wifi/index.html#wifi.Radio.connect does not describe the `timeout` argument. Also, it does not mention that DHCP is used by default and how it actually works.

### Additional information

Originally reported on https://forums.adafruit.com/viewtopic.php?p=959677

stanelie, 2024-12-03:
I'm having the same issue. Timeout parameter appears to be ignored.
Because of this blocking, I can't fallback on a static IP if DHCP fails.

stanelie, 2024-12-04:
I just ran another test : 

When I try to connect my ESP32-S3-zero to a wifi that has the DHCP disabled with `wifi.radio.connect(SSID, PASSWORD, timeout=5)`, it just sits there, without timing out. But then, I enable the DHCP server on the access point AFTER the timeout period has expired, and I receive a `WiFi connection timed out.` message in REPL!

So, some of the timeout function is working, somewhere...