adafruit/circuitpython #7561: edit 'settings.toml' from a RPI by GEANY or NANO bricks the PICO_W

kllsamui, 2023-02-09:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0 on 2023-02-06; Raspberry Pi Pico W with rp2040
Board ID:raspberry_pi_pico_w
```

### Code/REPL

```python
not available
```

### Behavior

when edit 'settings.toml'
directly on the CIRCUIPY drive
using Raspberry Pi OS
editors
- - - GEANY
- - - NANO

drive vanish & never show up again ( also not under windows )

( must nuke and CP800 again )

### Description

the default 'recommended software' Raspberry Pi  for PICO W Circuit Python
is the MU editor,
but can only install 1.0.3 there,
what not allow to edit a settings.toml file at all
//
on PC can install MU editor 1.2.0
where a manually typed 'settings.toml' allow to edit it there,
BUT NOT the original created 0 long file!!!
only a overwritten copy from PC can be edited later

### Additional information

but using RPI THONNY works

or edit on RPI and copy to CIRCUITPY drive also works

tannewt, 2023-02-10:
Are you pressing the reset button after editing the file? Those editors may not sync immediately and leave a bad file system.

kllsamui, 2023-02-10:
@tannewt 
no, no RESET connected / used

and it not looks like accidental, the PICO_W ( with CP800 ), more like bricked 4 times in a row with GEANY
2 people same project different continents 
use / 3 RPI / 3 PICO / several USB cable / 2 win PC ( for check if CIRCUITPY drive come up there later? but NO )

at save of 'settings.toml' PICO should just restart like with every file-change ( *.py usually done by MU editor )

inhowfar that file is treated differently ( from code.py... )

and what is the problem with the original 0 long / Mu 1.2.0 can not open?

dhalbert, 2023-02-10:
Geany and Nano are both known to _not_ flush the whole file out immediately to CIRCUITPY, and are disrecommended for that reason. See https://learn.adafruit.com/welcome-to-circuitpython/recommended-editors#editors-that-are-not-recommended-3104729, and read the whole page.

closed 2023-02-11 by kllsamui

kllsamui, 2023-02-11:
@dhalbert 
> Geany and Nano are both known to _not_ flush the whole file out immediately to CIRCUITPY, and are disrecommended for that reason.

so until there is  a MU editor on **RPI** what is able to edit 'settings.toml'

you recommend ? THONNY ?

or you say the file should be edited OFFLINE and copy to CIRCUITPY ?

resources show no info about that file handling:
https://docs.circuitpython.org/en/latest/docs/environment.html
https://learn.adafruit.com/scrolling-countdown-timer/create-your-settings-toml-file

dhalbert, 2023-02-11:
Some recommended editors are here: https://learn.adafruit.com/welcome-to-circuitpython/recommended-editors#recommended-editors-3104731. Can you use `gedit`? If it's not installed already on the RPi, you should be able to `apt install` it. VS Code is also fine, but is more complicated.

>or you say the file should be edited OFFLINE and copy to CIRCUITPY ?

That could also be done. When copying, do a `sync` immediately afterwards to insure the file was fully copied, e.g.
```sh
$ cp settings.toml /media/yourusername/CIRCUITPY; sync
```
On Linux, the `sync` command will force any unfinished writes out to all the disk drives, including CIRCUITPY.

Another idea (which I haven't tried) would be to edit `settings.toml` as `settings.py` with Mu, and then rename it to `settings.toml` from the command line or from the file browser.