adafruit/circuitpython #7562: PyPortal TFT left hand edge getting cutoff when using case with 8.0.0 release

caternuson, 2023-02-09:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0 on 2023-02-06; Adafruit PyPortal with samd51j20
```

### Code/REPL

```python
n/a
```

### Behavior

n/a

### Description

Left hand edge of display gets cutoff when using the PyPortal with the Desktop Stand case:
https://www.adafruit.com/product/4146

Looks OK with 7.3.3.

### Additional information

See related forum thread here:
https://forums.adafruit.com/viewtopic.php?p=959876

dhalbert, 2023-02-09:
This may be due to status bar (title bar) changes, maybe in #6473?

tannewt, 2023-02-10:
This was a deliberate change I made. We used to leave a blank column below blinka because the terminal's top left corner was to the right of it. Now, with 8.0, the title bar terminal is to the right of blinka and the regular terminal can now be positioned below it.

I don't know what we'd want to change in CP for this.

dhalbert, 2023-02-10:
Can the terminal TileGrid (is it a TileGrid), just be redone to be narrower and its origin repositioned? This could be done for the PyPortal in general, or we could offer a software solution (in boot.py or code.py?) for those using the case.

And maybe we should respin the case design.

PR #7880 (2023-04-20): Add PicoDVI support
PicoDVI in CP support 640x480 and 800x480 on Feather DVI, Pico and Pico W. 1 and 2 bit grayscale are full resolution. 8 and 16 bit color are half resolution.

Memory layout is modified to give the top most 4k of ram to the second core. Its MPU is used to prevent flash access after startup.

The port saved word is moved to a watchdog scratch register so that it doesn't get overwritten by other things in RAM.

Right align status bar and scroll area. This normally gives a few pixels of padding on the left hand side and improves the odds it is readable in a case. Fixes #7562

F [... cut]
commit 66edcf5d (2023-04-20): Add PicoDVI support

PicoDVI in CP support 640x480 and 800x480 on Feather DVI, Pico and
Pico W. 1 and 2 bit grayscale are full resolution. 8 and 16 bit
color are half resolution.

Memory layout is modified to give the top most 4k of ram to the
second core. Its MPU is used to prevent flash access after startup.

The port saved word is moved to a watchdog scratch register so that
it doesn't get overwritten by other things in RAM.

Right align status bar and scroll area. This normally gives a few
pixels of padding on the left hand side and improves the odds it is
readable in a case. Fixes #7562

 [... cut]
closed 2023-04-20 by dhalbert