adafruit/circuitpython #7565: Board freeze on generator.close()

gelsrc, 2023-02-10:
### CircuitPython version

```python
8.0.0 on Waveshare RP2040-Zero
8.0.0 on Luatos Core ESP32C3
```

### Code/REPL

```python
#
# This code freeze the board when run after second REPL restart / after code.py autoreload
#
def gen():
    while True:
        yield 0

while True:
    g = gen()
    print(next(g))
    g.close()
```

### Behavior

The bug occurs after restarting the Python machine. Just after reset / boot code works well. The second run freeze the board.

### Description

I checked this on two different board: Waveshare RP2040-Zero and Luatos Core ESP32C3. The bug is 100% reproduciable.

Some test results:

* 7.3.3 -- ok, no bug
* 8.0.0-beta.4 -- ok, no bug
* 8.0.0-beta.5 -- have a bug
* 8.0.0 -- have a bug

gelsrc, 2023-02-10:
It looks like the bug related with f3169246b, b499275bb.

When I build 8.0.0 with **MICROPY_CPYTHON_EXCEPTION_CHAIN=0** the freeze is gone.

jepler, 2023-02-10:
Thanks, that's excellent information @gelsrc !

jepler, 2023-02-10:
I don't think it's a micropython core issue, as they haven't (to my knowledge) taken the chained-exception patch, nor do they reset the VM the way we do

tannewt, 2023-02-13:
I meant our "micropython core". Maybe we need to use an upstream tag in addition to this one.

dhalbert, 2023-02-14:
Fixed by #7566.

PR #7566 (2023-02-10): Fix several places where an exception could be chained wrongly
If an exception's chain or context can refer to a pointer from a different VM, a crash would typically result.

This couldn't turn up on UNIX testing because the VM is never torn down and rebuilt like it is on hardware.

Testing performed: Ran the original reproducer on a Feather RP2040 Scorpio and verified the reported behavior. Ran with the fix and it no longer reproduced.

While I think that WatchdogException could have led to similar behavior, and made fixes for this, I didn't test it. I did verify scenarios with KeyboardInterr [... cut]
commit 0d957fe1 (2023-02-10): Fix several places where an exception could be chained wrongly

If an exception's chain or context can refer to a pointer from a different
VM, a crash would typically result.

This couldn't turn up on UNIX testing because the VM is never torn
down and rebuilt like it is on hardware.

Because in the 'static' case the GeneratorObject is now fully initialized
whenever it's raised, the initialization can be dropped, which reduces
the flash size slightly.
Closes: #7565