adafruit/circuitpython #7572: Code is reloaded when there is no write

Roming22, 2023-02-12:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0 on 2023-02-06; SparkFun Pro Micro RP2040 with rp2040
Board ID:sparkfun_pro_micro_rp2040
```

### Code/REPL

```python
> sudo fatrace -c
md5sum(141934): RO /run/media/user/KEYBOARD-L/code.py
md5sum(141934): C /run/media/user/KEYBOARD-L/code.py
```

### Behavior

```
Code stopped by auto-reload. Reloading soon.

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
```

### Additional information

I'm running on the Fedora 37 version.

Doing `find | LC_ALL=C sort | cpio -o | md5sum` on the CP drive before and after the reload shows the same checksum, showing that no files were modified.

There is a 0.5KB change in the `iostat` output, showing that maybe the OS tried to update metadata file (e.g. last modified data). If that's true, since the filesystem does not support that attribute, nothing has actually been written to disk even though a write action was performed.

dhalbert, 2023-02-13:
Could you explain the scenario? Did you do any writes slightly before this auto-reload? Or was the board quiescent for a long time?

Auto-reload is triggered when a write command comes down from USB. The code waits 0.75 seconds after the last write received before triggering an auto-reload, to allow multiple writes in quick succession.

So something is arriving from the host that is a write. The way to check this would be to use wireshark or similar to see what is actually arriving.

The FAT filesystem does support a last modified time. Could you elaborate on your comment _since the filesystem does not support that attribute_ ? Thanks.

What editor are you using? Some editors don't flush to disk immediately. Or are you using `cp` from the command line or similar? If the latter, it's helpful to follow the `cp` with an immediate `sync`. Linux can delay writing for 10 seconds or more sometimes.

Roming22, 2023-02-13:
I'm sorry, I meant "last accessed time", not "last modified". I've modified my original comment. I would expect the OS to try to update that metadata after each `open` on the file.

Minimally reproducible scenario:

- Unplug, re-plug the board.
- Open Mu and connect to serial console.
- The drive label has been changed to something else than "CIRCUITPY", so Mu is not able to interfere with the drive.
- Wait for a while: nothing happens.
- `cat /path/to/drive/code.py`.
- After a few seconds, the board auto-reloads.

dhalbert, 2023-02-13:
I tried to reproduce this on Ubuntu 22.04. My code.py is just the `print("Hello, world!")` you get on a fresh filesystem.

I did not rename the drive. I connect to the board either with Mu or with `tio /dev/ttyACM0`. I see "Auto-reload is on..." Nothing happens. I `cat /media/halbert/CIRCUITPY/code.py`. Nothing happens after that.

So something is happening on your particular system that is causing a write. Do you have any indexing programs or other scanning utilities running? Do you see this on other than Fedora? What is in your code.py? If you make it very simple, does the same thing happen?

RetiredWizard, 2023-02-13:
I tried to get this to happen from Debian 11 on my Sparkfun RP2040 Thingplus but rather than Mu, I used tio as the terminal. I'm not auto mounting the CIRCUITPY drive so I manually mounted after plugging in the board.

I couldn't get the reload to occur by just performing the cat /media/user/code.py but the board did reload when I perform the mount or unmount. I also noticed that the reload only occurred if I had a python script running. If I was at the REPL prompt mount/umount did not cause a reload.

RetiredWizard, 2023-02-13:
Okay, I just use cat on a larger file and while I was typing in that last message the board did an auto-reload. It was much more than a few seconds though, probably closer to a minute after I typed out the file..

RetiredWizard, 2023-02-13:
If I use the command `cat /media/user/filename.py; sync` the reload happens immediately, but only the first time I cat the file and only if a script is running.

Roming22, 2023-02-13:
My understanding is that you've reproduced the problem. If that's not correct, let me know and I'll try to provide an easy way to reproduce the issue.

dhalbert, 2023-02-13:
@Roming22 How large is your code.py, and was it running when you cat'd it?

dhalbert, 2023-02-13:
You can disable auto-reload, if it is interfering with your work:
```py
import supervisor
...
supervisor.runtime.autoreload = False
```

tannewt, 2023-02-15:
The md5sum won't consider file metadata. There may be writes to the FAT filesystem itself rather than the files in it. You could add a print within circuitpython if you are really curious about what block is being written to trigger the reload.

Roming22, 2023-02-15:
@tannewt Correct. Here's another test that hopefully shows that the metadata has not been modified as well.
```
> stat /run/media/rarnaud/KEYBOARD-R/* >before.txt; cat /run/media/rarnaud/KEYBOARD-R/code.py; time sleep infinity;
from time import sleep

print()
while True:
    print(end=".")
    sleep(2)
^C
real	0m37.156s
user	0m0.000s
sys	0m0.002s

17:31:39 - MBA-FEDORA:~
> stat /run/media/rarnaud/KEYBOARD-R/* >after.txt; diff before.txt after.txt 

```

How would I activate the logging you're talking about? Assuming I get some kind of address on the disk, what can I do with that information to move the troubleshooting forward?

dhalbert, 2023-02-15:
The metadata can be something invisible to you via the filesystem, such as the dirty bit. Or it may be that the host is doing a write that does not actually change a block.

The way to do the print is to be able to build CircuitPython yourself, and then insert an `mp_printf(&mp_plat_print, ...)` at the right place in the code. An alternative to this is to use wireshark and look at the USB traffic, which does not require a rebuild. wireshark I think will be able to parse the SCSI-like commands sent to the mSC device.

tannewt, 2023-02-16:
The relevant code is here: https://github.com/adafruit/circuitpython/blob/e47e6454e19986598960c3cd7b0f9d512545818a/supervisor/shared/usb/usb_msc_flash.c#L183-L221

Roming22, 2023-02-16:
I found that function while troubleshooting. I don’t know what to do with it though. 
1. If I add a print in that function, I guess I would have to recompile CP to troubleshoot?
2. If I can just add a decorator to the exported python function (under which module/name would I find it), I’ve no idea what to do with the info next.

dhalbert, 2023-02-17:
Here is a UF2 for youthat prints on every USB write:
[sparkfun_pro_micro_rp2040_usb_write_printing.uf2.zip](https://github.com/adafruit/circuitpython/files/10762198/sparkfun_pro_micro_rp2040_usb_write_printing.uf2.zip)
```c
int32_t tud_msc_write10_cb(uint8_t lun, uint32_t lba, uint32_t offset, uint8_t *buffer, uint32_t bufsize) {
    (void)lun;
    (void)offset;
    autoreload_suspend(AUTORELOAD_SUSPEND_USB);

    const uint32_t block_count = bufsize / MSC_FLASH_BLOCK_SIZE;

    fs_user_mount_t *vfs = get_vfs(lun);
    mp_printf(&mp_plat_print, "write lba=0x%x, block_count=%d\n", lba, block_count);     ######### added
    disk_write(vfs, buffer, lba, block_count);
```
I erased the filesystem with `storage.erase_filesystem()` and then reset the board. Here is some annotated output:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
write lba=0x1, block_count=1             #### writing some metadata

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 8.0.0-15-gbbadc0059-dirty on 2023-02-16; Adafruit Feather RP2040 with rp2040
>>> write lba=0x2, block_count=1     #### here I added one character to code.py
write lba=0x1f, block_count=1
write lba=0x41, block_count=1
write lba=0x1f, block_count=1
write lba=0x2, block_count=1
```
Let us know what it prints when it does an auto-reload.

dhalbert, 2023-02-17:
> 2\. If I can just add a decorator to the exported python function (under which module/name would I find it), I’ve no idea what to do with the info next.

There is no exported Python function to decorate, sorry, it's C code.

Roming22, 2023-02-17:
```
Code stopped by auto-reload. Reloading soon.

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

...............write lba=0x21, block_count=1

Code stopped by auto-reload. Reloading soon.
```

dhalbert, 2023-02-17:
It's hard to tell what is being written: it might be a directory entry or it might be a file. Try doing `storage.erase_filesystem()` so we can reproduce your results. Then `touch` code.py and see which lba is written. Then edit code.py slightly and do the same.

Roming22, 2023-02-17:
I erased the storage as requested. That was successful as I saw the label of the drive change.

When I add en empty line to code.py:
```
Code stopped by auto-reload. Reloading soon.

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

.......write lba=0x45, block_count=1

Code stopped by auto-reload. Reloading soon.
write lba=0x21, block_count=1
write lba=0x2, block_count=1
write lba=0x51, block_count=1
write lba=0x21, block_count=1

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

...write lba=0x2, block_count=1

Code stopped by auto-reload. Reloading soon.

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
```
When I touch or diff code.py, I get the single `lba=0x21`.

dhalbert, 2023-02-17:
I am doing the same as you, on Ubunut 22.04, and I don't see any writes.

I think `lba=0x21` may be directory metadata.

Are you running `tracker`? `tracker daemon -p` would say what `tracker` processes are running.

How long does it take before you see the `lba=0x21` write?

When you say `diff` the file, do you mean you were doing something like `diff /media/username/CIRCUITPY /dev/null`?

What does `mount -l |grep CIRCUITPY` say?

On mine:
```sh
$ mount |grep CIRCUITPY
/dev/sdb1 on /media/halbert/CIRCUITPY type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```

Roming22, 2023-02-17:
1. I don't know what tracker is but here's what I get:
```
> tracker3 daemon -p
Found 1 PID…
Found process ID 2441 for “tracker-miner-fs-3”
```
2. About 30-40s. Doing an actual write to the file does trigger the auto-reload immediately.
3. `diff --brief path/to/source/code.py /path/to/CIRCUITPY/code.py`.
4. 
```
> mount -l|grep CIRCUITPY
/dev/sdc1 on /run/media/rarnaud/CIRCUITPY type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2) [CIRCUITPY]
```

Roming22, 2023-02-17:
```
> gsettings list-recursively | grep -i org.freedesktop.Tracker | grep removable-devices
org.freedesktop.Tracker3.Miner.Files index-removable-devices false
```

Roming22, 2023-02-17:
I've added a `.nomedia` at the root of the drive which is supposed to disable tracker. I still get the auto-reloads.

Roming22, 2023-02-17:
Not sure if it means its indexed or not:
```
> tracker3 info /run/media/rarnaud/CIRCUITPY/code.py 
Querying information for entity: '/run/media/rarnaud/CIRCUITPY/code.py'
  No metadata available for that URI
Data object “/run/media/rarnaud/CIRCUITPY/code.py” currently exists
```

dhalbert, 2023-02-17:
When you run `fatrace`, do you see it report activity at the same time as the autoreload?

This is key and I should have paid more attention earlier.
> ```
> sudo fatrace -c
> md5sum(141934): RO /run/media/user/KEYBOARD-L/code.py
> md5sum(141934): C /run/media/user/KEYBOARD-L/code.py
> ```

Something is running `md5sum` on `code.py`. It might not be `tracker`, but it is something. Also, whenever something does a read on a file on CIRCUITPY, a write is done to note the access. And why would something run `md5sum` on code.py? To see if it changed, for some kind of indexing purposes

Since the `md5sum` is almost certainly transitory, we want to know who forked it. It looks like `sudo forkstat -d` would help with that.

Interestingly the Sparkfun Pro Micro has a 16MB external flash. That is bigger than  8MB I am testing with. What does`sudo fsck.fat -v -n /dev/sdb1` say? (substitute the correct `/dev/...` on your system)

I have:
```
 sudo fsck.fat -v -n /dev/sdb1
fsck.fat 4.2 (2021-01-31)
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "MSDOS5.0"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
      1024 bytes per cluster
         1 reserved sector
First FAT starts at byte 512 (sector 1)
         1 FATs, 16 bit entries
     14848 bytes per FAT (= 29 sectors)
Root directory starts at byte 15360 (sector 30)
       512 root directory entries
Data area starts at byte 31744 (sector 62)
      7137 data clusters (7308288 bytes)
63 sectors/track, 255 heads
         1 hidden sectors
     14336 sectors total
There is no label in boot sector, but there is volume label 'CIRCUITPY' stored in root directory
  Auto-copying volume label from root directory to boot sector.
Checking for unused clusters.
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.

Leaving filesystem unchanged.
/dev/sdb1: 9 files, 5/7137 clusters
```

dhalbert, 2023-02-17:
Do you have an editor or something running that might be checking to see if code.py was changed periodically? Could be a tab open on VSCode or something like that.

Roming22, 2023-02-17:
I'm the one running either `md5sum`, `diff`, or `cat`. So that call is not unexpected. `fatrace` logs the access right away, but as we can see in the logs I've provided, no write is done. The auto-reloads happens some time later (35-40s). Maybe when the system performs a `sync` on the device?

I'm running this because I'm working off a git repository and auto-uploading any changes on the file as I saved it.

Here's the disk info:
```
> sudo fsck.fat -v -n /dev/sdd1
fsck.fat 4.2 (2021-01-31)
Checking we can access the last sector of the filesystem
Boot sector contents:
System ID "MSDOS5.0"
Media byte 0xf8 (hard disk)
       512 bytes per logical sector
      2048 bytes per cluster
         1 reserved sector
First FAT starts at byte 512 (sector 1)
         1 FATs, 16 bit entries
     15872 bytes per FAT (= 31 sectors)
Root directory starts at byte 16384 (sector 32)
       512 root directory entries
Data area starts at byte 32768 (sector 64)
      7664 data clusters (15695872 bytes)
63 sectors/track, 255 heads
         1 hidden sectors
     30720 sectors total
Checking for unused clusters.
Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
 Automatically removing dirty bit.

Leaving filesystem unchanged.
/dev/sdd1: 80 files, 173/7664 clusters

```

dhalbert, 2023-02-17:
It could be that Fedora has a delayed `sync` for some reason compared with Ubuntu. I'm otherwise at a loss, since it's not reproducible for me under otherwise identical circumstances. I have been mentioning wireshark monitoring. Do you want help setting that up? wireshark is not going to tell you who did the USB transaction, though.

Roming22, 2023-02-17:
Sure :)

dhalbert, 2023-02-17:
1. Install wireshark with `dnf` or whatever (it's been a long time since I used Fedora).
2. Check that `usbmon` is in your kernel. Read https://fedoraproject.org/wiki/Usbmon and follow its instructions. It looks like it's already part of the kernel. On Ubuntu, I have to load it.
3. Add yourself to the `wireshark` and `usbmon` groups (per the link above). You will need to log out or even reboot to make this take effect, depending on how your desktop login works.
4. Start wireshark and look for the usb devices. It shows a little graph of activity, and you can see which usbmon channel is attached to which ports. There may be one that is all activity. Start capturing in wireshark. Plug in the board: it should generate a lot of activity. Then it will go quiet. You can filter the capture. You can also save the capture to pass it along.

.... hmm, rethinking ... Honestly, though I'm not sure what this is going to show, other than that something is doing a write. I was suggesting wireshark earlier to confirm the write, but my instrumented circuitpython did that. You really need to find out who is doing the write. Since `fatrace` is not showing that, then it may be the kernel.

If you do a `sync` a few seconds after the board is plugged in, does the auto-reload still happen much later, as before?

I may leave you to this, because it's really a Fedora problem. We can't do anything about it.

Roming22, 2023-02-17:
OK, I won't bother with wireshark then. I might play with sync to see if it changes the behavior, but I agree that the issue does not look like it's related to CP.

I'll just rsync the content of the drive locally, and diff with that, and copy the file when needed. That's good enough of a workaround for this weird situation.

dhalbert, 2023-02-17:
Thanks for your patience; closing.

closed 2023-02-17 by dhalbert

Roming22, 2023-02-17:
Thank you for your effort trying to get to the bottom of this!

Roming22, 2023-02-17:
As a final comment, if I run the `cat` and a `sync` immediately after, the auto-reload occurs right away.

dhalbert, 2023-02-17:
> As a final comment, if I run the `cat` and a `sync` immediately after, the auto-reload occurs right away.

interesting!

bill88t, 2023-02-17:
I remember as a mount option `noatime`. It's about not recording access time. It's default on rpi installations.

bill88t, 2023-02-17:
Perhaps ubuntu has it in the `defaults` mount flags while fedora doesn't.

Roming22, 2023-02-18:
@bill88t You legend!

`sudo mount -o remount,noatime /path/to/drive` solved the issue :)

dhalbert, 2023-02-18:
Both `mount` descriptions, for Fedora and Ubuntu, above have `relatime`, which is supposed to suppress writing about plain accesses, but not modifications. But glad this solves the problem for you!