adafruit/circuitpython #7583: MQTT connect is not working with circuitpython kernel with Jupyter Notebook

sean-bayarea, 2023-02-15:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0 firmware, Raspberry Pi Pico W with rp2040
```

### Code/REPL

```python
import os
import time
import ssl
import socketpool
import wifi
import adafruit_minimqtt.adafruit_minimqtt as MQTT

try:
    # Localhost broker address and port
    MQTT_BROKER = '192.168.0.42'
    MQTT_PORT = 1883

    # WiFi network configuration
    WIFI_SSID = 'my-wifi-ssid'
    WIFI_PASSWORD = 'my-wifi-password'

    # Define the callback functions
    def on_connect(client, userdata, flags, rc):
        print('Connected to MQTT broker')

    def on_disconnect(client, userdata, rc):
        print('Disconnected from MQTT broker')

    def on_message(client, topic, message):
        print('Received message on topic {0}: {1}'.format(topic, message))

    # Initialize the WiFi interface
    wifi_radio = wifi.radio
    wifi_radio.connect(WIFI_SSID, WIFI_PASSWORD)
    
    pool = socketpool.SocketPool(wifi.radio)

    # Initialize the MQTT client
    mqtt_client = MQTT.MQTT(
        broker=MQTT_BROKER,
        port=MQTT_PORT,
        is_ssl=False,
        username=None,
        password=None,
        socket_pool=pool,
    )

    # Set the callback functions
    mqtt_client.on_connect = on_connect
    mqtt_client.on_disconnect = on_disconnect
    mqtt_client.on_message = on_message

    # Connect to the MQTT broker
    mqtt_client.connect()

    # Subscribe to a topic
    mqtt_client.subscribe('test')

    # Publish a message to the topic
    mqtt_client.publish('test', 'Hello, MQTT!')

    # Wait for messages
    while True:
        mqtt_client.loop()
        time.sleep(0.01)
except Exception as e:
    print("Error:\n", str(e))
    print("Resetting microcontroller in 10 seconds")
    time.sleep(10)
    microcontroller.reset()
```

### Behavior

 I am using circuitpython 8.0.0 firmware. The following code works in Thonny with Pico W without any issue. But when I move this code in Jupyter notebook environment with circuitpython kernel, it failed at mqtt_client.connect(). The error message is "MMQTTException: No data received from broker for 10 seconds."

### Description

This error only happened in Jupyter notebook environment with circuitpython kernel.

tannewt, 2023-02-15:
Does other code work with the jupyter kernel? 8.0.0 added status bar updates from circuitpython that the kernel may not know to ignore. (Thonny does)

sean-bayarea, 2023-02-16:
Hi, Scott

Sample code for lighting up the system light is working in Jupiter
Notebook. Anything related to network has issue, for example,
adafruit_request is not working in Jupyter Notebook too.

Sean

tannewt, 2023-02-16:
Could you post the jupyter file as well? Thanks!

sean-bayarea, 2023-02-16:
Hi,

Please see the attachment. I included both mqtt and adafruit_requests samples.

for code session 1-7, it runs perfectly, the light is blinking for Pico W.

But mqtt and adafruit_requests sessions both are not working.

[CPX_blink_Pico_v2v.zip](https://github.com/adafruit/circuitpython/files/10760669/CPX_blink_Pico_v2v.zip)

tannewt, 2023-02-17:
@brentru You know mqtt and jupyter. Any ideas?

jposada202020, 2023-03-01:
Hello, I did not test this with the original workflow. Instead I used the one provided in the https://github.com/adafruit/circuitpython_jupyter_kernel/blob/master/examples/using_magics_and_nbdev.ipynb example. Instructions for installation are provided. I used the following script fromthis learning guide: https://learn.adafruit.com/mqtt-in-circuitpython/connecting-to-the-adafruit-io-mqtt-broker

```python

# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

import time
import ssl
import socketpool
import wifi
import adafruit_minimqtt.adafruit_minimqtt as MQTT

# Add a secrets.py to your filesystem that has a dictionary called secrets with "ssid" and
# "password" keys with your WiFi credentials. DO NOT share that file or commit it into Git or other
# source control.
# pylint: disable=no-name-in-module,wrong-import-order
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

# Set your Adafruit IO Username and Key in secrets.py
# (visit io.adafruit.com if you need to create an account,
# or if you need your Adafruit IO key.)
aio_username = secrets["aio_username"]
aio_key = secrets["aio_key"]

print("Connecting to %s" % secrets["ssid"])
wifi.radio.connect(secrets["ssid"], secrets["password"])
print("Connected to %s!" % secrets["ssid"])
### Feeds ###

# Setup a feed named 'photocell' for publishing to a feed
photocell_feed = secrets["aio_username"] + "/feeds/photocell"

# Setup a feed named 'onoff' for subscribing to changes
onoff_feed = secrets["aio_username"] + "/feeds/onoff"

### Code ###

# Define callback methods which are called when events occur
# pylint: disable=unused-argument, redefined-outer-name
def connected(client, userdata, flags, rc):
    # This function will be called when the client is connected
    # successfully to the broker.
    print("Connected to Adafruit IO! Listening for topic changes on %s" % onoff_feed)
    # Subscribe to all changes on the onoff_feed.
    client.subscribe(onoff_feed)

def disconnected(client, userdata, rc):
    # This method is called when the client is disconnected
    print("Disconnected from Adafruit IO!")

def message(client, topic, message):
    # This method is called when a topic the client is subscribed to
    # has a new message.
    print("New message on topic {0}: {1}".format(topic, message))

# Create a socket pool
pool = socketpool.SocketPool(wifi.radio)

# Set up a MiniMQTT Client
mqtt_client = MQTT.MQTT(
    broker="io.adafruit.com",
    port=1883,
    username=secrets["aio_username"],
    password=secrets["aio_key"],
    socket_pool=pool,
    ssl_context=ssl.create_default_context(),
)

# Setup the callback methods above
mqtt_client.on_connect = connected
mqtt_client.on_disconnect = disconnected
mqtt_client.on_message = message

# Connect the client to the MQTT broker.
print("Connecting to Adafruit IO...")
mqtt_client.connect()

photocell_val = 0
for i in range(4):
    # Poll the message queue
    mqtt_client.loop()

    # Send a new message
    print("Sending photocell value: %d..." % photocell_val)
    mqtt_client.publish(photocell_feed, photocell_val)
    print("Sent!")
    photocell_val += 1
    time.sleep(5)
```

Result

[image]

Notes:
For the printout to show in the Jupyter as probably witnessed already you will needto wait for the sctipt to finish running.
Not sure if this is of much help, but maybe you could try this workflow and see if this work for your needs

btw this was tested on:
```python

Adafruit CircuitPython 8.0.0 on 2023-02-06; Adafruit Feather ESP32-S2 TFT with ESP32S2
 

```

brentru, 2023-04-20:
@sean-bayarea I've tested the connection issue of this example without problems on a Pico W running CircuitPython 8.0.5 and the latest MiniMQTT library. I am connecting to the HiveMQ public MQTT broker.:

[image]

Below is a copy of the ipynb file you uploaded earlier. At the bottom, I've updated your code snippet a bit to move the `try/except` inside the `while True:` loop.:

[CPX_blink_Pico_v2v (fixed).ipynb.zip](https://github.com/adafruit/circuitpython/files/11288638/CPX_blink_Pico_v2v.fixed.ipynb.zip)

However,

> The error message is "MMQTTException: No data received from broker for 10 seconds."

You may try increasing the `keep_alive` kwarg within the broker object's initialization and see if that fixes it from your end. You are using a local broker, so I am unable to reproduce this portion of your test exactly: https://docs.circuitpython.org/projects/minimqtt/en/latest/api.html#adafruit_minimqtt.adafruit_minimqtt.MQTT

@dhalbert @tannewt This can be closed.

closed 2023-06-21 by jepler