adafruit/circuitpython #7588: Feather 2040 neopixel off by default or lower brightness

adriangalilea, 2023-02-15:
Hello,

I hope I'm not being nitpicky, I appreciate all the work that you do and love your products.

The feather 2040 board neopixel is too bright, it has that kind of led intensity that it can create a bright spot in your vision for an hour, I don't think that can be good for the vision long term without any diffuser.

And this wouldn't be a problem if it weren't on by default right as you flash it and run hello world.

It is also inconvenient that in order to turn it off I have to explicitly download the neopixel library and then use:

`import neopixel
pixel = neopixel.NeoPixel(board.NEOPIXEL, 1)
pixel.brightness = 0`

On my code.py

But the part that I hated the most is to realize that when I unplug it from my pc with the lipo plugged in, it again turns on despite having set the brightness to 0.

it's a bit frustrating and I'm typing this with a blind spot in the center of my vision. Lol

I'm sure I'm not the only one,

Thanks,

Adrian.

bill88t, 2023-02-15:
No you are not the only one who feels like neopixels are brighter than the sun.
Honestly they should never be used with a value > 5/255.
My eyes are sensitive to light and I work in pure darkness.
Night turns to day once I plug a neopixel board.
And god forbid I open the repl.
I use the `waveshare_rp2040_zero`.

tannewt, 2023-02-15:
You can change it from Python code with [`supervisor.runtime.rgb_status_brightness = 0`](https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime.rgb_status_brightness)

With 8.0.0 the LED should only blink unless you are in the REPL. [Default brightness is 50](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/status_leds.c#L56)

Neradoc, 2023-02-15:
I think we can look at changing the default brightness. It is 63 for neopixels, 50 for dotstars. (Out of 255)
It should remain visible in well lit environments by default though.
Maybe we can also change the REPL behavior ?

jposada202020, 2023-02-15:
my two cents, it is bright, If I need to look to a pin silk forget it.. I need to put my finger in the LED. btw 90% of the time I work in the REPL :) But, just a disclaimer I need glasses to see small things

RetiredWizard, 2023-02-15:
As a short term work around:

```python
import board
import neopixel_write
import digitalio

pin = digitalio.DigitalInOut(board.NEOPIXEL)
pin.direction = digitalio.Direction.OUTPUT
pixel_off = bytearray([0, 0, 0])
neopixel_write.neopixel_write(pin, pixel_off)
```

dhalbert, 2023-02-15:
```py
import supervisor
supervisor.led_status_brightness = 0
```
will turn it off (as mentioned by @tannewt above).
https://docs.circuitpython.org/en/latest/shared-bindings/supervisor/index.html#supervisor.Runtime.rgb_status_brightness

adriangalilea, 2023-02-15:
> It should remain visible in well lit environments by default though.

I'm guessing the white led signals that the board is working correctly.

But it is very bright, specially the white which is all at the same time. So I would either suggest lowering the brightness significantly or having that as an optional documented feature that you could enable easily rather than blinding unexpected people right after the flashing reboot sequence.

tannewt, 2023-02-16:
@adriangalilea Please post a video of what you are seeing. There's a chance that the neopixel is misbehaving.

adriangalilea, 2023-03-01:
> @adriangalilea Please post a video of what you are seeing. There's a chance that the neopixel is misbehaving.

Sorry for the delay, I've been meaning to do it, will do tomorrow.

catequalsgood, 2023-05-11:
While I personally do not have a huge problem with the current brightness of the status LED in general, I would really appreciate the ability to control the brightness during boot specifically. In my opinion the flashing in different colors can appear a little erratic to someone who does not know what the device is doing. A simple
 
```python
import supervisor
supervisor.runtime.rgb_status_brightness = 0
```

in `boot.py` does not seem to effect brightness during boot. (Which makes sense)

bill88t, 2023-05-11:
This would currently require a custom build to work.
It could *maybe* be made to read a `settings.toml` value.

It's a 'maybe' cause those flashes are *really* early (iirc in reset_port).
If the filesystem has init'ed by then, sure it's 2 lines of code.

Still that would be an api change, and would probably be held off until the next major release.
I am not familiar with the procedure for this.

It does sound nice having the ability to disable the led flashes entirely though. Not only for neopixels.

dhalbert, 2023-05-12:
The initial colors mostly use `0x30` (48) as the color value:
https://github.com/adafruit/circuitpython/blob/99a70474846c38348d4e91c74262e3959f642905/supervisor/shared/rgb_led_colors.h#L5-L15

Then this is further scaled by the default brightness values in
https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/status_leds.c
https://github.com/adafruit/circuitpython/blob/99a70474846c38348d4e91c74262e3959f642905/supervisor/shared/status_leds.c#L42-L43
https://github.com/adafruit/circuitpython/blob/99a70474846c38348d4e91c74262e3959f642905/supervisor/shared/status_leds.c#L55-56
If you are bothered by the current brightness levels, quantitatively, what would you suggest for your comfort _and_ for the average user (not operating in the dark)? The flashes need to be visible in a well-lit room.

Pushing this forward to 8.x.x.