adafruit/circuitpython #7596: Update CYW43 driver to latest

dhalbert, 2023-02-16:


bill88t, 2023-02-20:
When the merge happens I should retest picow-ap disconnection, should I not?

gneverov, 2023-02-20:
If we use the cyw43 driver from inside the pico sdk, #7595 would automatically update cyw43 driver.

dhalbert, 2023-03-07:
@jepler I have assigned myself #7595 and #7596. I will look at whether your changes to the cwy43 library are now incorporated. There are some interesting changes listed in the [1.5.0 release notes](https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.0) about cyw43:

Do you think these will make a difference?

>  Blocking cyw43_arch_wifi_connect_ functions now continue trying to connect rather than failing immediately if the network is not found.

> [restructuring of pico_cyw43_driver]

The updated cyw43 driver code does not quite compile. I'll look further at that.

PR #7697 (2023-03-09): pico-sdk to 1.5.0; cyw43-driver to 0.9.0
- Fixes #7595.
- Fixes #7596.

- Update pico-sdk to 1.5.0. There was some renaming of directories in the cyw43 part of the tree.
There is a new `pico_rand` API which we might use for `os.urandom` in the future. The network code is already using it. There is also a new "async contexts" API which could also be helpful. Right now these are compiled in only for CYW43 support.
- Update `cyw43-driver` to the latest version, and switching https://github.com/georgerobotics/cyw43-driver instead of our fork. Our fork had disabled mDNS by removing some code -- tha [... cut]
closed 2023-03-09 by tannewt