adafruit/circuitpython #7598: Separate Web Workflow and Auto Connect

Neradoc, 2023-02-17:
Web workflow requires auto-connect, using `CIRCUITPY_WIFI_SSID` and `CIRCUITPY_WIFI_PASSWORD`. These also automatically cause the webworkflow to start, web server on port 80, mdns and all. If `CIRCUITPY_WEB_API_PASSWORD` is not set, the home page shows up and cp/version.json, but the file browser and REPL return an empty page.

Auto-connect is useful, for development in particular, by not having to wait for reconnect on auto-reload (even though the user code should probably manage disconnects and reconnect attempts). But it would make sense for it to be disabled when `CIRCUITPY_WEB_API_PASSWORD` is not set. That way it doesn't take up a socket, port, and doesn't "leak" a mdns unintentionally.

We could also have a separate key to disable the web workflow instead, but it seems to me that people who only intend to use auto connect don't expect the web workflow to be running (despite the documentation mentioning it).

bill88t, 2023-02-21:
I have an idea, how about a `wifi.radio.bg_connect` that is enabled automatically when web workflow is configured and can be toggled at any time? Should this property be enabled, while backround tasks run, the wifi connection can be attempted automatically.

Though the more I think about it I think it should be a class..
```
wifi.radio.bg.connect
wifi.radio.bg.enabled
wifi.radio.bg.attempt_interval
```

tannewt, 2023-02-21:
It's ok to me to make web workflow use the API_PASSWORD to get going.

closed 2023-09-05 by tannewt