adafruit/circuitpython #7600: `help("modules")` duplicate entry for `ulab`

bill88t, 2023-02-17:
### CircuitPython version

```python
Adafruit CircuitPython 8.0.0-28-g8abce77971-dirty on 2023-02-17; Waveshare ESP32-S2-Pico with ESP32S2
```

### Code/REPL

```python
>>> help("modules")
__future__        digitalio         msgpack           sys
__main__          displayio         neopixel_write    terminalio
_asyncio          dualbank          nvm               time
_pixelmap         errno             onewireio         touchio
adafruit_bus_device                 espcamera         os                traceback
adafruit_bus_device.i2c_device      espidf            paralleldisplay   ulab
adafruit_bus_device.spi_device      espulp            ps2io             ulab
adafruit_pixelbuf fontio            pulseio           ulab.numpy
aesio             framebufferio     pwmio             ulab.numpy.fft
alarm             frequencyio       qrio              ulab.numpy.linalg
analogio          gc                rainbowio         ulab.scipy
array             getpass           random            ulab.scipy.linalg
atexit            gifio             re                ulab.scipy.optimize
audiobusio        hashlib           rgbmatrix         ulab.scipy.signal
audiocore         i2cperipheral     rotaryio          ulab.scipy.special
audiomixer        i2ctarget         rtc               ulab.utils
binascii          io                sdcardio          usb_cdc
bitbangio         ipaddress         select            usb_hid
bitmaptools       json              sharpdisplay      usb_midi
board             keypad            socketpool        uselect
builtins          math              ssl               vectorio
busio             mdns              storage           watchdog
canio             memorymap         struct            wifi
collections       microcontroller   supervisor        zlib
countio           micropython       synthio
Plus any modules on the filesystem
>>>
```

### Behavior

`ulab` is listed twice.

### Description

The build tagging is invalid, it's the settings-toml-pystack branch. Aka, master pretty much, so 8.0.2+.

jepler, 2023-02-17:
similar on
```
Adafruit CircuitPython 8.0.0-rc.2-6-gb03eed32f on 2023-02-03; Adafruit Feather ESP32-S2 TFT with ESP32S2
```

jepler, 2023-02-17:
This may be a CP-specific problem. I rebuilt micropython with ulab and `help("modules")` enabled, and get no duplicates. but, ulab submodules aren't listed either; mp treats submodules a bit differently than cp):
```
jepler@rat:~/src/micropython-ulab$ ./micropython/ports/unix/build-2/micropython-2 -c 'help("modules")'
__main__          mip/__main__      ucryptolib        ure
_thread           termios           uctypes           urequests
_uasyncio         uarray            uerrno            uselect
argparse          uasyncio/__init__ uhashlib          usocket
builtins          uasyncio/core     uheapq            ussl
cmath             uasyncio/event    uio               ustruct
framebuf          uasyncio/funcs    ujson             usys
gc                uasyncio/lock     ulab              utime
math              uasyncio/stream   umachine          utimeq
micropython       ubinascii         uos               uwebsocket
mip/__init__      ucollections      urandom           uzlib
Plus any modules on the filesystem
```
(`CFLAGS_EXTRA+=-DMICROPY_PY_BUILTINS_HELP=1 CFLAGS_EXTRA+=-DMICROPY_PY_BUILTINS
_HELP_MODULES=1`)

dhalbert, 2023-02-17:
`ulab` is in both `shared-bindings` and `extmod`. I haven't looked in detail but perhaps that is the reason.

PR #7601 (2023-02-17): Fix duplicate ulab help
Closes: #7600
commit d249af2a (2023-02-17): don't explicitly register ulab here

.. it is registered by MP_REGISTER_MODULE in ulab.

Closes: #7600
closed 2023-02-18 by MicroDev1