adafruit/circuitpython #7606: minimqtt lib problem with TLS broker

kllsamui, 2023-02-18:


DavePutz, 2023-02-21:
Testing (with a lot of tracing) has shown that the 116 is, in fact, an  MP_ETIMEDOUT error code  being returned from lwip_tcp_receive in common-hal/socketpool/Socket.c. I am unsure why the value is 116, as MP_ETIMEDOUT is defined as 110 in py/mperrno.h. Adding a "keep_alive=100" to the MQTT initialization in the supplied script avoids the timeout in my testing. There may well be some issue here that is keeping the error from being properly reported, perhaps because at some point the return value is being passed back up the chain as an unsigned.

vladak, 2023-02-21:
Nice work @DavePutz. For the record, the value of 115 matches `MP_EINPROGRESS` in `circuitpython/py/mperrno.h`, might be related considering the single zero byte returned from the function.

DavePutz, 2023-02-21:
One mystery solved; the #define of ETIMEDOUT in /arm-none-eabi/include/sys/errno.h is 116. Now the question is whether that error should be returned to the user, or perhaps a retry in the minimqtt library, or maybe increase the default value for TLS connections from 60 in the library. @kllsamui , are you able to test with your original setup and verify that a keep_alive of 100 is sufficient?

vladak, 2023-02-21:
Cool. I think in such case`recv_into` should raise an exception. The `_wait_for_msg()` will [deal with it](https://github.com/adafruit/Adafruit_CircuitPython_MiniMQTT/blob/342b8c99b2c8f10b605cc8311bf5f394f4d18e92/adafruit_minimqtt/adafruit_minimqtt.py#L967-L980):
```python

        # CPython socket module contains a timeout attribute
        if hasattr(self._socket_pool, "timeout"):
            try:
                res = self._sock_exact_recv(1)
            except self._socket_pool.timeout:
                return None
        else:  # socketpool, esp32spi
            try:
                res = self._sock_exact_recv(1)
            except OSError as error:
                if error.errno in (errno.ETIMEDOUT, errno.EAGAIN):
                    # raised by a socket timeout if 0 bytes were present
                    return None
                raise MMQTTException from error
```

kllsamui, 2023-02-22:
@DavePutz 
[image]

DavePutz, 2023-02-22:
@kllsamui , the value of 100 I used was based on the two MQTT brokers I had available to test (1 local, 1remote). Some sites could require even more time. The fix in PR #7623 should take care of the problem regardless of the keep_alive timeout.

kllsamui, 2023-02-22:
@DavePutz 

try 5000
[image]

so i have to test the ?next? CP802++daily from 
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberry_pi_pico_w/en_US/
( the newest NOW is 2023-02-21T22:26:38 )

or look for a new /lib/adafruit__minimqtt/ bundle?

kllsamui, 2023-02-22:
@dhalbert 
pls reopen

kllsamui, 2023-02-22:
use 'logging' again

[image]

why it say INSECURE
```
INFO - Establishing an INSECURE connection to <xxx>.hivemq.cloud:8883
```

kllsamui, 2023-02-22:
```
1615.053: DEBUG - waiting for messages for 0 seconds
1615.560: DEBUG - waiting for messages for 0 seconds
1616.066: DEBUG - waiting for messages for 0 seconds
1616.572: DEBUG - waiting for messages for 0 seconds
```
is meant to say 0.5 sec ?

vladak, 2023-02-22:
INSECURE means plaintext MQTT. If you want TLS, you need to set is_ssl to True.

> 
> 
> use 'logging' again
> 
> image <https://user-images.githubusercontent.com/19226581/220561096-fe6878a3-f814-4a16-afc3-fb47a4db2934.png>
> 
> why it say INSECURE
> 
> `INFO - Establishing an INSECURE connection to <xxx>.hivemq.cloud:8883
`

vladak, 2023-02-22:
I think that's because default loop() timeout is 0. See issue 142 in the Adafruit MQTT project.

> 
> 
> `1615.053: DEBUG - waiting for messages for 0 seconds
> 1615.560: DEBUG - waiting for messages for 0 seconds
> 1616.066: DEBUG - waiting for messages for 0 seconds
> 1616.572: DEBUG - waiting for messages for 0 seconds
`

jepler, 2023-02-22:
<details>
<summary>OP's testing code</summary>

```python
# ______________________________________________________ TEST CODE for LOCAL REMOTE broker TLS problem

import time
import microcontroller
#import micropython
#import gc # micropython garbage collection # use gc.mem_free() # use gc.collect

import os # ____________________________________________ expect file: settings.toml with:
THIS_REVISION = os.getenv('THIS_REVISION')
THIS_OS = os.getenv('THIS_OS')

WIFI_SSID = os.getenv('WIFI_SSID')
WIFI_PASSWORD = os.getenv('WIFI_PASSWORD')

WIFI_IP = os.getenv('WIFI_IP')

DIAG = True # False # ___________________________________ global print disable switch / overwritten by console [D][enter]
DIAG = bool(os.getenv('DIAG')) # ________________________ now get from settings.toml

def dp(line=" ", ende="\n"):
    if DIAG : print(line, end=ende)

import socketpool
from ipaddress import ip_address
import wifi

from adafruit_httpserver.server import HTTPServer
from adafruit_httpserver.request import HTTPRequest
from adafruit_httpserver.response import HTTPResponse
from adafruit_httpserver.methods import HTTPMethod
from adafruit_httpserver.mime_type import MIMEType

dp("___ Connecting to router {:s} OR CHECK THE 'settings.toml' FILE".format( WIFI_SSID) )
wifi.radio.set_ipv4_address( # _______________________ fixIP ( requires > CP 8.0.0 beta 4 )
    ipv4=ip_address(WIFI_IP),
    netmask=ip_address("255.0.0.0"),
    gateway=ip_address("192.168.1.1"),
    ipv4_dns=ip_address("192.168.1.1"),
)
wifi.radio.connect(WIFI_SSID, WIFI_PASSWORD)
dp("___ Connected to {:s}".format( WIFI_SSID) )

dp("___ Listening on http://{:s}:80 ".format(str(wifi.radio.ipv4_address)) )

pool = socketpool.SocketPool(wifi.radio)

# make a WEB SERVER
server = HTTPServer(pool)

HtmlStyle =   '<style>'
HtmlStyle +=   'html {font-family: "Times New Roman", Times, serif; background-color: lightgreen;'
HtmlStyle +=   'display:inline-block; margin: 0px auto; text-align: center;}'
HtmlStyle +=   'h1{color: deeppink; width: 200; word-wrap: break-word; padding: 2vh; font-size: 35px;}'
HtmlStyle +=   'p{font-size: 1.5rem; width: 200; word-wrap: break-word;}'
HtmlStyle +=   '.button{font-family: {font_family};display: inline-block;'
HtmlStyle +=   'background-color: black; border: none;'
HtmlStyle +=   'border-radius: 4px; color: white; padding: 16px 40px;'
HtmlStyle +=   'text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}'
HtmlStyle +=   'p.dotted {margin: auto; width: 75%; font-size: 25px; text-align: center;}'
HtmlStyle +=   '</style>'

def html_index_page():
    htmldata =  '<!DOCTYPE html><html><head><title>KLL Pico W Web Server</title>'
    htmldata += HtmlStyle
    htmldata += '</head><body>'
    htmldata += '<h1>KLL Pico W Web Server</h1><h2> from Circuit Python'
[... 136 lines cut ...]
    response = HTTPResponse(request)
    with response:
        response.send(body=html_index_page(), content_type=MIMEType.TYPE_HTML)

server.start(str(wifi.radio.ipv4_address)) # ______________ startup the server

mqtt_connect() # __________________________________________ above MQTT start NOTE: broker must be up first

while True :
    try:
        server.poll()
    except OSError:
        print("ERROR server poll")

    try:
        mqtt_client.loop() # ______________________________ now we see the subscribed tuning come through
    except Exception as e:
        print("Error: MQTT loop\n", str(e))
        MQTTok = False
        #time.sleep(10) # __________________________________ broker reboot expected
        #microcontroller.reset() # _________________________ try reboot case broker recovered

    time.sleep(0.5)
```
</details>

kllsamui, 2023-02-23:
no news, just for info:
* nuke
* adafruit-circuitpython-raspberry_pi_pico_w-en_US-20230222-c2c7b93.uf2
* adafruit-circuitpython-bundle-8.x-mpy-20230223.zip

works with ```is_ssl=True,```
HANGS without
```
___ Connecting to router kll-wlan_2.4G OR CHECK THE 'settings.toml' FILE
___ Connected to kll-wlan_2.4G
___ Listening on http://192.168.1.214:80 
___+++ setup MQTT
___+++ setup MQTTclient
___++++ use TLS
___ Attempting to <xxx>.hivemq.cloud
```
after several minutes wait see
```
Error: MQTT connect
 Repeated connect failures
___ Publishing to PICO_W
Hello Broker: i am a PICO W
Error: MQTT publish hello
 MiniMQTT is not connected
___ Subscribing to PICO_W/set tuning
Error: MQTT subscribe tuning
 MiniMQTT is not connected
___ MQTT broker:<xxx>hivemq.cloud
___ mqtt_topic: PICO_W
Error: MQTT loop
 Unable to receive 1 bytes within 60 seconds.
Error: MQTT loop
 MiniMQTT is not connected
```
_______________
with logging :
```
___ Connecting to router kll-wlan_2.4G OR CHECK THE 'settings.toml' FILE
___ Connected to kll-wlan_2.4G
___ Listening on http://192.168.1.214:80 
___+++ setup MQTT
___+++ setup MQTTclient
___++++ use TLS
___ Attempting to connect to <xxx>hivemq.cloud
1207.934: DEBUG - Attempting to connect to MQTT broker (attempt #0)
1207.938: DEBUG - Attempting to establish MQTT connection...
1207.943: INFO - Establishing an INSECURE connection to <xxx>.hivemq.cloud:8883
1208.250: DEBUG - Sending CONNECT to broker...
1208.254: DEBUG - Fixed Header: bytearray(b'\x10-\x00')
1208.259: DEBUG - Variable Header: bytearray(b'\x04MQTT\x04\xc2\x00<')
1208.264: DEBUG - Receiving CONNACK packet from broker
1268.272: INFO - MMQT error: Unable to receive 1 bytes within 60 seconds.
1268.277: DEBUG - Reconnect timeout computed to 2.00
1268.281: DEBUG - adding jitter 0.10 to 2.00 seconds
1268.286: DEBUG - Attempting to connect to MQTT broker (attempt #1)
1268.290: DEBUG - Attempting to establish MQTT connection...
1268.293: DEBUG - Sleeping for 2.1 seconds due to connect back-off
1270.399: INFO - Establishing an INSECURE connection to <xxx>hivemq.cloud:8883
1270.707: DEBUG - Sending CONNECT to broker...
1270.711: DEBUG - Fixed Header: bytearray(b'\x10-\x00')
1270.715: DEBUG - Variable Header: bytearray(b'\x04MQTT\x04\xc2\x00<')
1270.720: DEBUG - Receiving CONNACK packet from broker
1330.729: INFO - MMQT error: Unable to receive 1 bytes within 60 seconds.
1330.734: DEBUG - Reconnect timeout computed to 4.00
1330.738: DEBUG - adding jitter 0.39 to 4.00 seconds
1330.742: DEBUG - Attempting to connect to MQTT broker (attempt #2)
1330.746: DEBUG - Attempting to establish MQTT connection...
1330.750: DEBUG - Sleeping for 4.39 seconds due to connect back-off
1335.145: INFO - Establishing an INSECURE connection to <xxx>hivemq.cloud:8883
1335.453: DEBUG - Sending CONNECT to broker...
[... 74 lines cut ...]
___ Flags: 0 RC: 0
2645.850: DEBUG - Resetting reconnect backoff
___ Publishing to PICO_W
Hello Broker: i am a PICO W
2645.857: DEBUG - Sending PUBLISH
Topic: PICO_W
Msg: b'Hello Broker: i am a PICO W'                            
QoS: 0
Retain? False
___ Published to PICO_W with PID 0 
___ Subscribing to PICO_W/set tuning
2645.873: DEBUG - SUBSCRIBING to topic PICO_W/set with QoS 0
___ Subscribed to PICO_W/set with QOS level 0
___ MQTT broker:<xxx>hivemq.cloud
___ mqtt_topic: PICO_W
2646.419: DEBUG - waiting for messages for 0 seconds
2646.925: DEBUG - waiting for messages for 0 seconds
2647.433: DEBUG - waiting for messages for 0 seconds
```

pls see that actually in booth cases there is a
```
Receiving CONNACK packet from broker
```
i am confused...

dhalbert, 2023-02-23:
We believe this is fixed by #7632.

closed 2023-02-23 by dhalbert

jepler, 2023-02-23:
from what I can see, hivemq only offers mqtt over tls (what we call is_ssl=True). It's no surprise if they can't successfully communicate when one side is expecting TLS and the other is expecting plaintext.

[image]

jepler, 2023-02-23:
Port 1883 is allocated for non-secure MQTT. However, hivemq machines do not respond on port 1883. I think that non-TLS (use_ssl=False) is not supported by hivemq.

berarduc, 2023-07-12:
Has this issue been fixed? I'm using latest 8.2 circuitpython and minimqtt library and this problem still occurs.

vladak, 2023-07-14:
> Has this issue been fixed? I'm using latest 8.2 circuitpython and minimqtt library and this problem still occurs.
> 

This issue is closed, so it would be better to open new one, describe your problem in detail and reference this issue.

PR #7623 (2023-02-21): Correctly raise OS error in socketpool_socket_recv_into()
Fixes #7606. Instead of returning an errno (which seems to lose the negative sign along the way and so will be interpreted as a received count) instead raise an OSerror with the errno in socketpool_socket_recv_into(). Tested using TLS with a local mosquitto server and io.adafruit.com.
PR #7632 (2023-02-22): raspberrypi: SSLSocket: raise OSError when appropriate
Rather than returning the negative error value.

This is intended to close #7606, ~~though I did not test with mqtt~~ (EDIT by @dhalbert: see second post). Instead, I created a simple standalone test program:
```python
import wifi, socketpool, ssl, time
#wifi.radio.connect(<omitted>)
import socketpool
socket = socketpool.SocketPool(wifi.radio)
ctx = ssl.create_default_context()

b = bytearray(8)

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sss = ctx.wrap_socket(s, server_hostname='example.com')
sss.connect(('example.com', 443))
s [... cut]
commit de9233f8 (2023-02-22): raspberrypi: SSLSocket: raise OSError when appropriate

Rather than returning the negative error value.

This is intended to close #7606, though I did not test with mqtt.
Instead, I created a simple standalone test program:
```python
import wifi, socketpool, ssl, time
#wifi.radio.connect(<omitted>)
import socketpool
socket = socketpool.SocketPool(wifi.radio)
ctx = ssl.create_default_context()

b = bytearray(8)

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
sss = ctx.wrap_socket(s, server_hostname='example.com')
sss.connect(('example.com', 443))
sss.setblocking(False)
r = sss.recv_into( [... cut]